ArduPilot/ardupilot #6926: No observable messages when waypoint delay starts

Evidlo, 2017-09-09:
#### Issue details
The `MAV_CMD_NAV_WAYPOINT` command has an optional `delay` option that makes the vehicle wait inside the waypoint for the given number of seconds.  Right now, the MISSION_ITEM_REACHED message triggers only at the end of this delay right before the drone begins to move towards the next waypoint.

There needs to be an additional message (perhaps `MISSION_ITEM_LOITER`) sent right after the vehicle stops so that users who need to trigger an action at standstill (like a long-exposure camera or some sort of sensor) can do so asynchronously.
#### Version
APM:Copter V3.6-dev (d91f89d2)

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

OXINARF, 2017-09-11:
Why don't you use a normal waypoint and then a nav delay command?

Evidlo, 2017-09-12:
That works too.  I suppose the downside is that you can only store half the waypoints this way.

Is there proper support for reading commands from an sdcard yet?

OXINARF, 2017-09-12:
> That works too. I suppose the downside is that the maximum waypoints is cut in half.

That's true, but I feel like that is the correct approach to what you want to achieve. Adding another message that is sent mid-command seems like a hack to me.
By the way, you should aim to have all actions be done by the autopilot; MAVLink is lossy protocol, a message can be lost in transit.

> Is there proper support for reading commands from an sdcard yet?

No, SD card is too slow to read from during a flight.

closed 2021-03-17 by Evidlo