ArduPilot/ardupilot #6934: [rover] arming via switch in skid track rover

andresiko, 2017-09-11:
#### Issue details
armong via switch in a skid track/tank rover can result in a dangerous spin, disabling this shall be mandatory, and implementing a switch arming could be interesting.

#### Version
3.1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[*] Rover
[  ] Submarine

#### Airframe type
"big" skid steering rover

#### Hardware type
pixhawk 2

#### Logs
_Please provide a link to any relevant logs that show the issue_

gmorph, 2017-09-12:
Not nearly enough information to go on.  What exactly did you do in great detail please?  Can you provide a telemetry or dataflash log of when this issue is occurring?

andresiko, 2017-09-25:
Sorry, I am not sure on how to procceed in this forum, and i didn't want to flood...
Arming via transmitter stick in a tank is quite dangerous, because you are ordering the tank to run fast backwards while turning.
@rmackay9  considered adequate to [allow disarming only in hold mode](https://discuss.ardupilot.org/t/pivoting-speed-parameter-arming-via-transmitter/20522/16). But also a switch arming could be a good solution.

auturgy, 2017-10-15:
@peterbarker could your switch arming patch be pushed to Rover?

peterbarker, 2017-10-17:
@auturgy Yes, absolutely.

I haven't moved to do that as I'm hoping to get https://github.com/ArduPilot/ardupilot/pull/6819/ in and base it off that. There are *lots* of options in Copter's CH7_OPT stuff that should be common to all vehicles.

gmorph, 2017-12-17:
The CH7_OPT stuff has been enhanced to allow Arming/Disarming via a switch.  It will be in the next rover release.  I believe that resolves this issue so I'm closing out.

andresiko, 2017-12-17:
It is still a problem when channel 7 is already taken for other purpouse... Not possible to choose an arming switch? sorry for summoning this again!

gmorph, 2017-12-18:
You are correct that if CH7 is being used for something else then obviously you can't use it to arm/disarm.
We do have a PR which has been submitted which provides the framework of what your after - 
https://github.com/ArduPilot/ardupilot/pull/6819
whilst that's only for copter it puts in all the building blocks and then subsequent adding to Rover is trivial.  Its a big PR and I am getting through it but these things take time.
So this issue is technically solved by the CH7 switch and as the extra functionality your after is coming soon are you ok if I close this out?

andresiko, 2017-12-20:
I am absolutely OK with anything you decide! I don't know if my requests are possible to carry out, and I can't make an idea of how much effort it takes, if you consider that this issue is solved in any way then just close it, but it will be really nice to have the solution in the ardupilot/rover/docs.
Thank you again!

rmackay9, 2017-12-20:
I think this is all done in Rover-3.2 which is in beta testing now.  I've also documented the auxiliary switch features on this new wiki page: http://ardupilot.org/rover/docs/rover-auxiliary-functions.html

So i'm going to go ahead and close this one.

closed 2017-12-20 by rmackay9