ArduPilot/ardupilot #6935: [Rover] 3.2 skid steering bugged

andresiko, 2017-09-11:
#### Issue details
skid steering works fine in 3.1, but looks broken in 3.2 beta, This [video](https://www.youtube.com/watch?v=TpWiKL_9x7I) shows both 3.1 and 3.2 mission execution, and only 3.1 version do pivot. 
Param files can be found here:
[3.1](https://discuss.ardupilot.org/uploads/default/original/2X/4/4555d7452845b0e67e4b62c27c919c3572362b0d.param)
[3.2](https://discuss.ardupilot.org/uploads/default/original/2X/5/572e02927c4dbacf41fded4934ece6dcc881ae3e.param)

#### Version
3.1 & 3.2
#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[*] Rover
[  ] Submarine

#### Airframe type
"big" skid steering/tank rover

#### Hardware type
pixhawk 2

#### Logs

OXINARF, 2017-09-11:
This issues list is for feature requests and confirmed bugs. For user support please post in the forum (http://discuss.ardupilot.org), together with logs. This applies even more specially for beta versions, like 3.2 is.

khancyr, 2017-09-11:
@OXINARF it is confirmed see https://discuss.ardupilot.org/t/pivoting-speed-parameter-arming-via-transmitter/20522/9

andresiko, 2017-09-18:
I'm available for testing any progress! just update the beta and I will do any testing you request.

(if this posts is not appropriate just delete)

gmorph, 2017-12-17:
Can you please retest with the latest BETA or off master - I believe this issue is now resolved.  Let us know how you go.  Thanks.

rmackay9, 2018-01-17:
This should all be resolved now that the official Rover-3.2.0 release is out so closing.

closed 2018-01-17 by rmackay9