ArduPilot/ardupilot #6946: Once per second frame overrun introduced since Copter 3.5.2

priseborough, 2017-09-13:
#### Summary

When logging replay data using ArduCopter master at 58dec08 it was noticed that once per second the EKF IMU input is running 3 msec late.  

[image]

The same parameters and hardware setup on Copter 3.5.2 stable gives:

[image]

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
N/A - bench test of autopilot

#### Hardware type
Pixracer with dual GPS. 

#### Parameters

[param.txt](https://github.com/ArduPilot/ardupilot/files/1298422/param.txt)

OXINARF, 2017-09-13:
Copter 3.5.2 is quite far away from master, we should figure out what is wrong, but it's possible we'll need to narrow the commit range.

amilcarlucas, 2017-10-10:
Is this maybee the reason why PH2.1 enters "rodeo mode" when navigating in spline waypoints ?

I have [log of PH2.1 misbehaving on spline waypoints](https://github.com/ArduPilot/ardupilot/files/1333590/00000050.log) . It should always point (yaw) at the middle of the circle, but it instead it flyes fine for the first 535 seconds and after that tuns 90° off center. And @rmackay9 thinks that is not related to #6633

peterbarker, 2021-07-30:
We're definitely not getting this any more.  Closing.

closed 2021-07-30 by peterbarker