ArduPilot/ardupilot #6953: Quad ArduPlane 3.8.1 / SITL -- Floating point exception

expntly, 2017-09-14:
#### Issue details
SITL crashes right when I try to take-off with a quadplane.

```shell
$ uname -a
Darwin 15.6.0 Darwin Kernel Version 15.6.0: Sun Jun  4 21:43:07 PDT 2017; root:xnu-3248.70.3~1/RELEASE_X86_64 x86_64
$ git --version
git version 2.6.2
$ vagrant --version
Vagrant 2.0.0
$ VirtualBox --help
Oracle VM VirtualBox Manager 5.1.26
...

$ git clone https://github.com/ArduPilot/ardupilot.git ardupilot-3.8.1
$ cd ardupilot-3.8.1
$ git checkout tags/ArduPlane-3.8.1
$ vagrant up
$ vagrant ssh
...
$ git submodule update --init --recursive
$ exit
$ vagrant ssh -c "sim_vehicle.py -j 2 -v ArduPlane -f quadplane"
Ardupilot environment ready.  Run 'sim_vehicle.py' to start simulating an arducopter instance.
or run 'make sitl' to just do a test build.
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at -35.363261,149.165230,584,353 (CMAC)
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/vagrant/modules/waf/waf-light" "configure" "--board" "sitl"
Setting top to                           : /vagrant 
Setting out to                           : /vagrant/build 
Autoconfiguration                        : enabled 
Setting board to                         : sitl 
Checking for 'g++' (C++ compiler)        : /usr/bin/g++ 
Checking for 'gcc' (C compiler)          : /usr/bin/gcc 
Checking for need to link with librt     : not necessary 
Checking for HAVE_CMATH_ISFINITE         : yes 
Checking for HAVE_CMATH_ISINF            : yes 
Checking for HAVE_CMATH_ISNAN            : yes 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes 
Checking for header endian.h                   : yes 
Checking for header byteswap.h                 : yes 
Checking for program 'python'                  : /usr/bin/python 
Checking for python version >= 2.7.0           : 2.7.12 
Checking for program 'python'                  : /usr/bin/python 
Checking for python version >= 2.7.0           : 2.7.12 
Source is git repository                       : yes 
Update submodules                              : yes 
Checking for program 'git'                     : /usr/bin/git 
Checking for program 'size'                    : /usr/bin/size 
Benchmarks                                     : disabled 
Unit tests                                     : enabled 
Checking for program 'rsync'                   : /usr/bin/rsync 
'configure' finished successfully (1.455s)
SIM_VEHICLE: Building
SIM_VEHICLE: "/vagrant/modules/waf/waf-light" "build" "--target" "bin/arduplane" "-j" "2"
Waf: Entering directory `/vagrant/build/sitl'
.........
.........
[754/755] Linking build/sitl/lib/libArduPlane_libs.a
[755/755] Linking build/sitl/bin/arduplane
Waf: Leaving directory `/vagrant/build/sitl'

BUILD SUMMARY
Build directory: /vagrant/build/sitl
Target         Text     Data  BSS    Total  
--------------------------------------------
bin/arduplane  1528538   974  34496  1564008

Build commands will be stored in build/sitl/compile_commands.json
'build' finished successfully (32.743s)
SIM_VEHICLE: Using defaults from (/vagrant/Tools/autotest/default_params/quadplane.parm)
SIM_VEHICLE: Run ArduPlane
SIM_VEHICLE: "/vagrant/Tools/autotest/run_in_terminal_window.sh" "ArduPlane" "/vagrant/build/sitl/bin/arduplane" "-S" "-I0" "--home" "-35.363261,149.165230,584,353" "--model" "quadplane" "--speedup" "1" "--defaults" "/vagrant/Tools/autotest/default_params/quadplane.parm"
SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: "mavproxy.py" "--master" "tcp:127.0.0.1:5760" "--sitl" "127.0.0.1:5501" "--out" "10.0.2.2:14550" "--out" "10.0.2.2:14551"
RiTW: Starting ArduPlane : /vagrant/build/sitl/bin/arduplane -S -I0 --home -35.363261,149.165230,584,353 --model quadplane --speedup 1 --defaults /vagrant/Tools/autotest/default_params/quadplane.parm
RiTW: Window access not found, logging to /tmp/ArduPlane.log
Connect tcp:127.0.0.1:5760 source_system=255
Log Directory: 
Telemetry log: mav.tlog
MAV> Waiting for heartbeat from tcp:127.0.0.1:5760

Init ArduPlane V3.8.1 (d8b8b1ee)

Free RAM: 131072
Firmware change: erasing EEPROM...
done.
load_all took 0us
0 0 0 validate_structures:176: Validating structures
DataFlash_File: buffer size=16384
online system 1
INITIALISING> Mode INITIALISING
APM: ArduPlane V3.8.1 (d8b8b1ee)
Received 1006 parameters
Saved 1006 parameters to mav.parm
APM: Barometer calibration complete
APM: Airspeed calibration started
APM: Ground start complete
FBWA> APM: Airspeed sensor calibrated
Mode FBWA
APM: GPS 1: detected as u-blox at 115200 baud
APM: EKF2 IMU0 initial yaw alignment complete
APM: EKF2 IMU1 initial yaw alignment complete
APM: EKF yaw reset -15.77
APM: Init HOME
APM: EKF2 IMU0 tilt alignment complete
APM: EKF2 IMU1 tilt alignment complete
APM: EKF2 IMU0 Origin set to GPS
APM: EKF2 IMU1 Origin set to GPS
APM: EKF2 IMU0 is using GPS
APM: EKF2 IMU1 is using GPS
no link
link 1 down
link 1 OK
heartbeat OK
FBWA> 
FBWA> qstabilize
FBWA> Got MAVLink msg: COMMAND_ACK {command : 11, result : 0}
QSTABILIZE> Mode QSTABILIZE

QSTABILIZE> 
QSTABILIZE> param set Q_ENABLE
QSTABILIZE> Q_ENABLE         1.000000

QSTABILIZE> 
QSTABILIZE> arm throttle
QSTABILIZE> APM: Throttle armed
Got MAVLink msg: COMMAND_ACK {command : 400, result : 0}
QSTABILIZE> rc 3 1300
QSTABILIZE> /vagrant/Tools/autotest/run_in_terminal_window.sh: line 54:  6926 Aborted                 (core dumped) $cmd $* &> "$filename" < /dev/null

QSTABILIZE> 
QSTABILIZE> no link
link 1 down
SIM_VEHICLE: Keyboard Interrupt received ...
SIM_VEHICLE: Killing tasks
Connection to 127.0.0.1 closed.

$ vagrant ssh
Welcome to Ubuntu 16.10 (GNU/Linux 4.8.0-59-generic i686)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage

  Get cloud support with Ubuntu Advantage Cloud Guest:
    http://www.ubuntu.com/business/services/cloud

0 packages can be updated.
0 updates are security updates.

Your Ubuntu release is not supported anymore.
For upgrade information, please visit:
http://www.ubuntu.com/releaseendoflife

New release '17.04' available.
Run 'do-release-upgrade' to upgrade to it.

Last login: Thu Sep 14 16:28:10 2017 from 10.0.2.2
Ardupilot environment ready.  Run 'sim_vehicle.py' to start simulating an arducopter instance.
or run 'make sitl' to just do a test build.
ubuntu@ubuntu-yakkety:/vagrant/ArduCopter$ ls /tmp/
ArduCopter.log  ArduPlane.log
ubuntu@ubuntu-yakkety:/vagrant/ArduCopter$ cat /tmp/ArduPlane.log 
Started model quadplane at -35.363261,149.165230,584,353 at speed 1.0
Starting sketch 'ArduPlane'
Starting SITL input
Using Irlock at port : 9005
bind port 5760 for 0
Serial port 0 on TCP port 5760
Waiting for connection ....
Loaded defaults from /vagrant/Tools/autotest/default_params/quadplane.parm
bind port 5762 for 2
Serial port 2 on TCP port 5762
bind port 5763 for 3
Serial port 3 on TCP port 5763
Smoothing reset at 0.001
Loaded defaults from /vagrant/Tools/autotest/default_params/quadplane.parm
ERROR: Floating point exception - aborting

```

#### Version
ArduPlane 3.8.1, but I've seen this happen for a while.

The above works fine if I try to load copter and take-off.

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Default quad params.

peterbarker, 2017-09-15:
On Thu, 14 Sep 2017, expntly wrote:

> SITL crashes right when I try to take-off with a quadplane.

I believe I fixed this last week with 
85ebe923b663a9c7e2cc367fe303c007eeaace07

Please verify....

OXINARF, 2017-09-15:
If @peterbarker's thought is verified this is a duplicate of #6245.

expntly, 2017-09-15:
Confirmed, works now, thanks both!

closed 2017-09-15 by expntly