ArduPilot/ardupilot #6956: Helicopter: Changing Mode Too Soon Destroys Helicopter

R-Lefebvre, 2017-09-15:
If pilots starts motor, and takes off in Acro or Stabilize, then before the Rotor Runup timer is complete, the user changes to Auto, Guided, AutoTune, Brake, Circle, Drift, Sport or RTL the helicopter immediately goes to full negative collective and crashes downward violently.

R-Lefebvre, 2017-09-15:
Fix is probably this, but it is untested:

https://github.com/R-Lefebvre/ardupilot/commit/3b2871ccaf36e0b7a9bb34d47123c332aa978525

These protections are in AC3.3, but didn't make it into AC3.4/5.

tridge, 2017-09-15:
> These protections are in AC3.3, but didn't make it into AC3.4/5.

oh? I don't see them in 3.3.2

R-Lefebvre, 2017-09-15:
https://github.com/ArduPilot/ardupilot/blob/Copter-3.3/ArduCopter/control_auto.cpp#L28

R-Lefebvre, 2017-09-15:
This is the PR to 3.3

https://github.com/ArduPilot/ardupilot/commit/05ff279a18065a7527ea90f5240946ac9de8de8d#diff-a990cecda769fb510ae5be975034717b

OXINARF, 2017-09-15:
This was PRed against the 3.3 branch but never against master, which is why it's not in any release after.
Thankfully we are now doing backports and not really accepting PRs against specific branches, so I hope it doesn't happen again.

tridge, 2017-09-20:
I've fixed this in #6978

tridge, 2017-09-20:
closing as this is fixed in master now, and is in list of PRs to go into 3.5 (with Randys approval hopefully)
thanks for opening this issue Rob!

closed 2017-09-20 by tridge