ArduPilot/ardupilot #6957: QuadPlane: THR_SLEWRATE not working as documented

expntly, 2017-09-15:
#### Issue details

Documentation says:

> THR_SLEWRATE: Throttle slew rate
> maximum percentage change in throttle per second. A setting of 10 means to not change the throttle by more than 10% of the full throttle range in one second.

I performed some tests with SITL.

Here's THR_SLEWRATE=100:

[image]

Takes 1.2 seconds to ramp up to max throttle, as expected.

And here's THR_SLEWRATE=1

[image]

Takes 3.3 seconds to ramp up to max throttle, not as expected.

Some params:
> RC3_MAX          2000.000000
> RC3_MIN          1000.000000
> THR_MAX          100.000000
> THR_MIN          0.000000

I was expecting the second test to take 100 seconds to go from 1000 PWM to 2000 PWM, 10 PWM at a time.

#### Version
Plane 3.7

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
QuadPlane in SITL.

#### Logs

DF log with THR_SLEWRATE=1%:  https://www.dropbox.com/s/0uu4q5qmnbrk9bb/9.BIN?dl=0
DF log with THR_SLEWRATE=100%: https://www.dropbox.com/s/rgoo79qjef2n90w/10.BIN?dl=0

expntly, 2017-10-04:
Any update?  Thanks

tridge, 2017-10-04:
there is a lower limit of 1 PWM unit per loop. This is because the PWM value is stored as an integer, so if we allowed for a change of less than 1 then it would never change at all (as it would get rounded down).
On a quadplane the loop rate is 300Hz by default, so slew rate min is 300 PWM per second, which means 3.3 seconds for 1000 PWM.
We should update the docs, but its not actually a bug

tridge, 2017-11-12:
I've added a note on this to the docs

closed 2017-11-12 by tridge

kimkim999x, 2020-12-10:
hi, i have got a smimlar issue, I am testing my IC Engine on the ground, when I modify the parameter THR_SLEWRATE to 20 and test it with manual and FBWA, it doesn’t seems to slow down at all and still reach idle/full throttle at a split of a second, when I test it with SITL its doing the same…am I setting some parameters wrong or these flight mode does not support slew? if so how should I test the suitable slewrate…, thanks for helping

peterbarker, 2020-12-11:
@kimkim999x do you have a dataflash log of that happening?

vierfuffzig, 2020-12-12:
@kimkim999x @peterbarker THR_SLEWRATE is applied in auto throttle modes only

kimkim999x, 2020-12-12:
@peterbarker i am currently still doing it on SITL.. Will try it on the actual thing
@vierfuffzig Thanks for the advice and it is working indeed!!! but may i ask how will you test the right Slewrate for the IC engine indeed, with FBWB mode?