ArduPilot/ardupilot #6964: Servo Test Function Doesn't Work

R-Lefebvre, 2017-09-17:
AC3.3 had a servo test function that operated just after boot-up.  This exercised all the servos through their range.  Swashplate moved through it's full range of collective pitch and rolled the cyclic around.  This appears to be at least partially broken since AC3.4.  The collective function doesn't operate.

tridge, 2017-09-20:
can you explain a bit more how this worked? Did it always do the swash movement, or was there a parameter to turn it on? How did it interact with the safety switch? if the safety is on at startup then its pointless to try doing servo movement, so did it do the test when the safety was first disabled?

ChristopherOlson, 2017-09-23:
The parameter that enables or disables the servo check function is H_SV_TEST. Set to 0 it disables it. Set to any other value it runs the number of test cycles the parameter is set to.

In AC3.3.3 it used to first move the swashplate to H_COL_MIN then do one complete roll of the swashplate "cyclic ring" to the range set by the parameter H_CYC_MAX. It then moved the swashplate to the value set by H_COL_MAX and once again did a "cyclic ring" roll to the range set by H_CYC_MAX. It then moved to H_COL_MID, and this comprised one cycle of the test function.

In AC3.4 and later it does the "cyclic ring" but it does not move the collective to MIN and MAX.

The purpose of the function is to automatically exercise the servos on boot-up, bypassing the PID controller, so the pilot can observe for proper servo operation at the extreme of the collective and cyclic settings. This verifies the servos are operating correctly and that there is no binding in the mechancal linkages. In AC3.4 and later, as mentioned, the test function operates the cyclic ring range, but the pilot must have his/her radio by the aircraft while it is exercising the servos and manually move the collective pitch from min to max while the test function is running.

I hope this helps explain what it used to do.

ChristopherOlson, 2017-11-05:
This issue has been fixed by PR #7149 and merged to Copter master. So closing this.

R-Lefebvre, 2018-09-11:
Issue still partially exist.  The servo test function used to move the collective range smoothly and slowly, and now it jumps instantaneously.

bnsgeyer, 2018-09-20:
@R-Lefebvre Ok I see where I went wrong when I fixed this problem the first time.  I'll take care of it.

bnsgeyer, 2018-09-24:
PR #9467 was merged which fixed this issue.

closed 2018-09-24 by bnsgeyer