ArduPilot/ardupilot #6969: Notify LED broken

jaxxzer, 2017-09-18:
On px4-v2/ pixhawk 1 Notify LED is non-functional for Plane, Copter, Sub. This commit appears to be the culprit. 697131e94adb588dee895cb74b9e7ca07e5d3698

ns-1m, 2017-09-25:
This bug kills the switch LED cable and USB LED cable in pixhawk 1. I don't know if this bug is related to the PPM but it also stop my FlySky FS-i6s working in my new X8. I use 3.5.2 in my X4 (working) before I converted to X8.

jaxxzer, 2017-10-11:
This appears to be resolved in master. Notify led is working again. I do not know about @ns-1m's comment.

closed 2017-10-11 by jaxxzer

jaxxzer, 2017-10-11:
It looks to me like arming switch is working on 3.5.3 @ns-1m.

ns-1m, 2017-10-11:
OK. Notify LED might be fixed. But I got PX4IO problem (ie. PX4IO not found) with Pixhawk 1 ver 2.4.7 using AC 3.5.3 and AC 3.6 using MP and QGroundControl. PX4 1.6.5 (latest version of PX4) is working in this board using QGroundControl.

jaxxzer, 2017-10-11:
Ok, I suggest opening another issue describing exactly how you are determining there is an issue, because it is still not clear to me what your problem is. Instead of 'ie. PX4IO not found' put the exact error message you see and where/when you see it.

ns-1m, 2017-10-12:
OK. It seems that AC 3.5.3-r1 fixed the problem, able to load the firmware, connect to MavLink and got the calibration screen in MP. I uploaded the AC 3.5.3-r1 firmware using QGroundControl in Mac OS.