ArduPilot/ardupilot #6974: Arduplane - Quadplane RTL not reaching completion during mission

nick2204, 2017-09-19:
#### Issue details
After running an auto mission takeoff-> several waypoints -> RTL, the vehicle never re-enters the landed state.

When reattempting to arm in Guided mode, throttle goes to 100%

#### Version
Arduplane 3.8.2-dev (running in SITL)
Q_enable =1 (quadplane)
Q_guided_mode = 0 (FW loiter)
Q_RTL_mode = 1 (land in VTOL once RTL reached)

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Quadplane

#### Hardware type
SITL

#### Logs

[00000007.zip](https://github.com/ArduPilot/ardupilot/files/1313010/00000007.zip)

nick2204, 2017-09-19:
Also checked that doing RTL in the guided state, does not get it stuck in this state.

tridge, 2017-09-22:
I'm having difficulty reproducing this issue in SITL. I've used your parameters and your mission and it doesn't show any problems.
Could we worth through this together sometime so I can reproduce?

nick2204, 2017-09-22:
Sure I'll give you a Skype call Monday if you're free.

nick2204, 2017-09-27:
An extension to preventing some undesired behaviours for particular Quadplane airframes, it would be great to be able to:

1. Disable standard Fixed-wing takeoff for air frames with no wheels - prevent accidental forward motor arming on the ground

2. Disable navigation (other than position hold at home) within some 3D bounding box of arming location - prevent dragging landing gear sideways across the ground

3. Disable use of guided waypoints until after the vehicle has taken-off - prevent sudden movement as vehicle leaves the ground

magicrub, 2017-09-27:
@nick2204 those are all good ideas. Could you make 3 new issues so we can track those feature requests? Bonus points if you paste the URLs here for reference :)

nick2204, 2017-10-09:
@magicrub done :)
https://github.com/ArduPilot/ardupilot/issues/7048
https://github.com/ArduPilot/ardupilot/issues/7049
https://github.com/ArduPilot/ardupilot/issues/7050

DavidIngraham, 2018-02-07:
@nick2204 Are you still seeing this issue?

nick2204, 2018-02-07:
hey @DavidIngraham , most of the problems were fixed by @tridge in 3.8.3
at least i never saw the problem again after this.
The major change that prevented some of this behaviour was the addition of Q_options:
[Q_options](http://ardupilot.org/plane/docs/parameters.html#q-options-quadplane-options)

IamPete1, 2021-07-25:
seems to have been fixed?

closed 2021-07-25 by IamPete1