ArduPilot/ardupilot #6977: Overly sensitive swashplate response to cyclic inputs for Dual heli frame

bnsgeyer, 2017-09-19:
#### Issue details
A dual heli frame was added to Copter 3.5 to support tandem rotor helicopters.  It was noted in @frizensami post (https://discuss.ardupilot.org/t/tandem-helicopter-pid-tuning/17849) that tuning parameters for PID in pitch and roll seemed very low (1 order of magnitude smaller) when compared to single main rotor /tail rotor configurations.  Recent, @PittRBM posted (https://discuss.ardupilot.org/t/dual-heli-frame-frame-class-11-problem/20972) an issue when setting up his transverse configuration using the 3.5 version dual heli frame that the swashplate was overly sensitive using the tuning parameters that he used in tuning his personal version of a copter 3.4 dual heli frame derived from the same user @fhedberg.  
In comparing the motorsheli_single.cpp and the motorsheli_dual.cpp, I noticed an inconsistency in the code between the two files that I believe is causing this increased sensitivity.  
In these lines of code there is a term that is divided by 0.45 
https://github.com/ArduPilot/ardupilot/blob/528d4d83d70d26dbca0f1e7e786a32a5a107ec7b/libraries/AP_Motors/AP_MotorsHeli_Dual.cpp#L530-L535

where in the Motorsheli_Single.cpp file the same expression is multiplied by 0.45.
https://github.com/ArduPilot/ardupilot/blob/528d4d83d70d26dbca0f1e7e786a32a5a107ec7b/libraries/AP_Motors/AP_MotorsHeli_Single.cpp#L444-L445

I believe changing the expression to have it multiplied by 0.45 as is done in the heli_single frame will fix the oversensitivity of the cyclic in the heli_dual frame

#### Version
latest Copter 3.5.X

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
helicoter tandem and transverse

#### Hardware type
Pixhawk

#### Logs
No logs

tridge, 2017-10-02:
you're absolutely right, thanks! I've created a PR for it here:
https://github.com/ArduPilot/ardupilot/pull/7026

auturgy, 2017-10-15:
Can this be closed?

OXINARF, 2017-10-18:
Yep, fixed by #7026.

closed 2017-10-18 by OXINARF