ArduPilot/ardupilot #6988: Modify the default Parameter COMPASS_TYPEMASK = 0 to enable QMC5883 detection

pkocmoud, 2017-09-22:
#### Issue details
The Honeywell HMC5883/HMC5983 compasses are discontinued and hardware provider are having to switch to alternatives. The QMC5883 is such an alternative. In speaking with @tridge there was an earlier bug in the driver for this IC that resulted in the current default parameter COMPASS_TYPEMASK = 4096 that omitted scanning for the QMC5883. This driver bug has since been resolved. I have manually changed the parameter COMPASS_TYPEMASK = 0 and found no issues with the latest Master versions of ArduCopter and ArduPlane. 

Can we update the default parameter COMPASS_TYPEMASK = 0 allowing for all compasses to be detected and enabled?

#### Version
MASTER

#### Platform
[ ? ] All
[  ] AntennaTracker
[ X ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
NA

#### Hardware type
PixRacer - Tested

#### Logs
NA

OXINARF, 2017-09-23:
This is done #6764. I agree that it shouldn't be a problem to enable it, but I need to get approval to merge. I've added it to next dev call.

OXINARF, 2017-09-27:
Fixed by 6e286a2e661701fc687f26388b842ec47d33159d

closed 2017-09-27 by OXINARF