ArduPilot/ardupilot #7005: RC Passthrough on Soft Reboot

deadhurricane, 2017-09-25:
From https://discuss.ardupilot.org/t/rc-passthrough-on-soft-reboot/21342

Hi All,

A potential safety concern we've noticed is that RC controls are passed through directly to the motors when a reboot command is sent through a GCS software. The following conditions were setup for the RC Passthrough to occur:
Pixhawk 2.1, RFD900+ Telemetry, Taranis X9D+ with X8R, MAVProxy GCS. Steps taken:
1. Connect Power to Pixhawk 2.1 
2. Wait for boot to complete
3. (Optional) Press in Safety switch or mavproxy: arm safetyoff)
4. (Optional) Arm throttle
5. Issue reboot from mavproxy or PREFLIGHT_REBOOT_SHUTDOWN from Mission Planner
6. Wiggle RC Sticks and throttle to observe passthrough behaviour
7. Once Pixhawk has finished booting the RC controls do no pass through anymore.

Is there any way to stop the RC Pass through on a soft reboot? Very dangerous if throttle were to accidentally be non-zero on the reboot command.

Shreyas

#### Version
This was replicated on 3.8.2 with Pixhawk 2.1 as well as 3.7.1 on Pixhawk 1. This behaviour does not occur if the entire pixhawk is power cycled manually by disconnected the power source.

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
X6 Flying Wing &  3 Channel Plane

#### Hardware type
Pixhawk 1 and Pixhawk 2.1

geofrancis, 2017-10-07:
I always though this behavour was a safety feature incase of flight controller reset mid flight, so it didnt lock you out of the controls.

kabturek, 2017-10-12:
it looks like a feature to me

geofrancis, 2017-10-12:
As far as i know this has been the normal behavour since the APM incase of flight controller crash you still have manual control.

JMare, 2017-10-13:
In this case, the motor can come live on the ground without the safety switch ever being pressed.
This seems undesirable to me.

I understand the reason the behavior exists.
I am not well familiar with the IO co-processor code, but would it be possible for the co-processor to tell the difference between a PREFLIGHT_REBOOT_SHUTDOWN and a FMU failure?

WickedShell, 2018-08-26:
Do you have a BRD_SAFETY_MASK set? With BRD_SAFETY_MASK setup this is an expected (and desired behavior). The outputs shouldn't be passed though if the safety mask hasn't been set.

IamPete1, 2021-07-25:
Seems to be expected? Closing pending a update from @deadhurricane

closed 2021-07-25 by IamPete1