ArduPilot/ardupilot #7006: Add support for CRSF

magicrub, 2017-09-25:
Add support for new data protocol CRSF.
http://www.team-blacksheep.com/products/prod:crossfire_tx

3x faster version of SBUS.

magicrub, 2017-09-25:
Here's a PR from BetaFlight adding support:
https://github.com/betaflight/betaflight/pull/4199

ufoDziner, 2019-10-03:
Just curious if this is going to get implemented.  Thanks.

rmackay9, 2019-10-03:
@ufoDziner, it's hard to say.  It just takes one developer being interested in doing the implementation for it to happen... sometimes that person appears immediate, sometimes it takes a while... Sometimes a company (like one of our Partners) sponsors a developer to do the work... we plan out the big stuff at our annual conference but not the smaller items so much.  This is probably too long an answer but..

auturgy, 2019-10-03:
I’ll tag @tridge as he has the hardware now.
It’s definitely on the list, but can’t predict a date.

ufoDziner, 2019-10-03:
Thanks you two!  It would be awesome to get it implemented.

Target0815, 2019-10-10:
Ardupilot doesn't lack various protocols, but it would be nice if you could keep up with the times and implement current developments.

Therefore +1 for the CRSF protocol.

CodingGhost, 2019-12-22:
If somebody that know the AP code well points me in the right direction, I could try it.

andyp1per, 2019-12-22:
I'm thinking of doing this when I have finished the runcam support. It won't be too dissimilar https://github.com/ArduPilot/ardupilot/pull/12810

CodingGhost, 2020-03-25:
@andyp1per, any ETA? I could help if you want, we need this! :)

andyp1per, 2020-03-25:
Sorry, I'm busy implementing SRXL2, will get to it eventually

andyp1per, 2020-05-02:
This is a start. https://github.com/ArduPilot/ardupilot/pull/14257 Lacking a protocol spec so can't do the VTX part and lacking hardware so can't test the RC part!

CodingGhost, 2020-05-02:
@andyp1per I will test it for you!! I have all the hardware. I can try to get the protocol spec also, I have some connections ;)

CodingGhost, 2020-05-02:
you can shoot me a message via email

jelkins8, 2020-05-05:
I can also test if needed.

closed 2020-07-07 by tridge