ArduPilot/ardupilot #7009: in TECS, cos(Phi) may be incorrectly used.

luweikxy, 2017-09-26:
https://github.com/ArduPilot/ardupilot/blob/eebef857f1a24a720a9fbdc712b6c050cd7eb191/libraries/AP_TECS/AP_TECS.cpp#L621

# 1. The first problem:
In TECS cosPhi is calculated as sqrtf((rotMat.a.y*rotMat.a.y) + (rotMat.b.y*rotMat.b.y));
and I calculated it by hands, the resulte is fellow which is nopt equal to cosPhi.
sqrtf((rotMat.a.y*rotMat.a.y) + (rotMat.b.y*rotMat.b.y)) = sqrtf(cosTheta^2 + sinPhi^2 + cosPhi^2)

# 2. second problem:
In the next line, the comment says that use (1/cos(bank angle) - 1) for increase thrust when turn, yes, it is right! but in the next code, it is 1.0f/(cosPhi * cosPhi) - 1.0f). So the problem is how to understand 1.0f/(cosPhi * cosPhi) - 1.0f) in code? is it right or wrong?

Thanks a lot!

priseborough, 2017-10-29:
Induced drag increases with load factor^2 so the code is correct

OXINARF, 2017-11-05:
Closing based on @priseborough's comment.

closed 2017-11-05 by OXINARF