ArduPilot/ardupilot #7010: not logging MODE on re-arm

tridge, 2017-09-26:
problem: arm, then disarm, then change mode, then arm again
it does not log the mode on the 2nd arm
needs fixing for plane 3.8.3 as this leads to very misleading logs

OXINARF, 2017-09-27:
Probably a simple fix can be done for Plane 3.8.3 but I think this is a larger problem. When we disarm and then re-arm we don't rewrite the initial messages - it's possible that, in that interim, mission, parameters, mode, etc. were changed. In my opinion the correct fix would be to rewrite the initial messages, except the FMT ones.

peterbarker, 2017-09-28:
Current thinking for Plane 3.8.3 is that we'll write out critical messages even when disarmed.

Sounds easy, right?

I've previously had a patch which added `LOG_BITMASK_DSRM` (or something...).  This would supplant the `LOG_DISARMED` argument.

tridge and I threw around the concept of writing out all of the initial messages.  Amusingly, you do have to worry about formats due to `Log_Write(format, ...)`

OXINARF, 2017-09-29:
> Current thinking for Plane 3.8.3 is that we'll write out critical messages even when disarmed.
>
> Sounds easy, right?

It doesn't and looks even harder than writing all startup messages again. But you know the DataFlash library much better than me.

> tridge and I threw around the concept of writing out all of the initial messages. Amusingly, you do have to worry about formats due to Log_Write(format, ...)

Ah I forgot about that. That means we can have logs now that are missing FMT messages, right?

peterbarker, 2017-09-29:
On Thu, 28 Sep 2017, Francisco Ferreira wrote:

> Ah I forgot about that. That means we can have logs now that are missing FMT
> messages, right?

Indeed.

OXINARF, 2017-10-30:
> Ah I forgot about that. That means we can have logs now that are missing FMT messages, right?

@peterbarker did @tridge's commit fix the issue above?

tridge, 2017-10-30:
yes, this is fixed with the commit to log critical msgs while disarmed, closing

closed 2017-10-30 by tridge