ArduPilot/ardupilot #7012: Motor start running when arduplane restart with mission planner

marstears, 2017-09-27:
#### Issue details
This issue appeared only when I set my twin motor plane with two separated throttle channel. For example, I use channel 3 with left motor and channel 7 with right motor, set SERVO3_FUNCTION to 73 and SERVO7_FUNCTIONU to 74. Under this situation the motor will start running with full throttle when I send a PREFLIGHT_REBOOT_SHUTDOWN command using mission planner. And I tried to use a Y cable to make two motor using one throttle channel(default channel 3 and SERVO3_FUNCTION to 70), this issue will never happen. 

#### Version
ArduPlane 3.8.1 and 3.8.2.

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[#] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
My Twin Dream(MTD) twin motor plane.

#### Hardware type
ProfiCNC Pixhawk 2.1 with cube.

#### Logs
_Please provide a link to any relevant logs that show the issue_

Rolf-G, 2017-09-30:
The problem also occurs with a pixhawk 1 (Arduplane 3.8.2).

Kelly-Foster, 2017-10-06:
I have a video of oscilloscope output during Pixhawk2.1 reboot on my twin motor setup, if seeing the problem live is helpful.

Kelly

magicrub, 2017-10-06:
@Kelly-Foster  logs are more helpful.

Kelly-Foster, 2017-10-06:
Can do.  Currently on other RC projects, but will get back to PH2.1 / MyTwinDream soon.

Kelly

magicrub, 2017-10-06:
>Does logging survive reboot?

Yes, it dumps everything onto the SDcard.

JMare, 2017-10-21:
This is likely related to #7005 
On reboot RC channel 7 (Probably either mapped to a switch or left at 1500) is passed through to output 7

kikislater, 2017-10-28:
Hi, because there is no log here. I could post some. 

[00000124.BIN.zip](https://github.com/ArduPilot/ardupilot/files/1423965/00000124.BIN.zip)

Don't know if this log is sufficient, what I do is the following : 
- First boot with powering with lipo
- Reboot through Mission Planner
- Waiting for satellite
- Arm

If not, I could provide other logs

Naterater, 2019-01-04:
I think this behavior is not a problem any more. We had certain issues with quadplanes, but that has been addressed in https://github.com/ArduPilot/ardupilot/issues/9741.

IamPete1, 2021-08-16:
Fixed by #18050 we now treat throttle left and right the same as main throttle and they all honor the safety mask.

closed 2021-08-16 by IamPete1