ArduPilot/ardupilot #7025: autotune not correctly tuning large multi rotor.

dturetsky, 2017-10-01:
#### Issue details
_Please describe the problem, or desired feature_
My high school's robotics team has built a large octocopter (16 inch props) that we use to compete in the AUVSI SUAS competition.  After upgrading to version 3.5 we tried to autotune.  We found online that you could start auto tune from position hold and did it that way.  After tuning the copter would toilet bowl or kind of swing whenever it was given input from the rc transmitter.  This happened on all the axis but was far worse on the yaw axis.  This problem almost led to a crash which would have cost of thousands of dollars.  When attempting a autonomous landing the copter tilted over on the ground cause damage to a few props.
#### Version
_What version was the issue encountered with_
3.5.2
#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Octocopter
#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
pixhawk
#### Logs
_Please provide a link to any relevant logs that show the issue_
This is an image of the bad autotune values we got.
[image]

JacksonUAS, 2017-10-02:
Sounds more like a compass issue than an autotune one

magicrub, 2017-10-02:
agreed. Toilet bowling is typically, but not always, a compass thing

dturetsky, 2017-10-02:
We autotuned again today with the max aggressiveness value which seemed to fix it, but after autotune was complete the manual controls were a little off.  Meaning that pushing forward on the pitch stick cause the copter to go left.  It was in simple mode which might also suggest that it was a compass thing.

OXINARF, 2017-10-02:
This issues list is for feature requests and confirmed bugs. For user support please post in the forum (http://discuss.ardupilot.org).

closed 2017-10-02 by OXINARF