ArduPilot/ardupilot #7027: EK2_ALT_SOURCE : Baro Fall Back not working

JacksonUAS, 2017-10-02:
#### Issue details
With EK2_ALT_SOURCE=2 (to allow for more accurate geotagging alt in mapping (RTK)) the aircraft will not fall back to the baro with no gps fix, and instead appears to fly with an accelerometer based altitude with 0 reported altitude. 

Log contains flight cycling between EK2_ALT_SOURCE=2 and 0 

#### Version
3.5 x (Tridge SBF message fix)

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Quad X

#### Hardware type
PH2.1, AsteRX-M UAS RTK, Here, Lightware SF11C (disabled through channel option for test) 

#### Logs
_Please provide a link to any relevant logs that show the issue_

[00000007.log](https://github.com/ArduPilot/ardupilot/files/1347661/00000007.log)

JacksonUAS, 2017-10-02:
Wondering if this issue is to do with SBF Auto config not working, and the params for GPS alt not being updated correctly when the GPS has no fix?

OXINARF, 2017-10-03:
@priseborough Can you look at this? It looks to me that the problem isn't that the EKF doesn't fallback to baro, but that the controller won't use the altitude in that case because of https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_NavEKF2/AP_NavEKF2_Control.cpp#L486 When getting posD, it will return false and so the controller (in this case Copter) won't use it as it is deemed as invalid.

JacksonUAS, 2017-10-03:
Cheers for looking at it, let me know if you need any testing done.

priseborough, 2017-10-03:
If the origin has not been set, then GPS altitude cannot be used by the EKF. Changing the EK2_ALT_SOURCE parameter in-flight has not been tested and could have unintended consequences. If baro reversion needs to be tested, then the GPS_USE parameter should be used instead. I will add a reboot required attribute to the EK2_ALT_SOURCE parameter.

JacksonUAS, 2017-10-03:
Was booted up with EK2_ALT_SOURCE = 2, flown (in althold) and then switched to EK2_ALT_SOURCE = 0 and flown, and repeated. 0 worked as per usual, 2 resulted in some excitement. It appears to behave fine with "hotswapping" with disarming/arming in between.

priseborough, 2017-10-03:
@JacksonUAS I've looked at the log you attached and neither GPS receiver has a 3D lock so either you have attached the wrong log or you were never testing the GPS alt hold performance.

priseborough, 2017-10-03:
The arming checks need to prevent you from taking off in alt_hold when the height is invalid.

JacksonUAS, 2017-10-03:
We were testing altitude hold with no GPS Fix (indoors) and EK2_ALT_SOURCE = 2 , aircraft flies fine with EK2_ALT_SOURCE = 2 with an RTK fix.

priseborough, 2017-10-03:
OK, that makes sense. 

The reason I am pushing back on changing the validity reporting is that the original use case for EK2_ALT_SOURCE = 2 was where the baro is so badly affected by ground effect that taking off in a height controlled mode with baro gave unacceptable height variation. Some of the large heavy lift copters can have 15m of baro height disturbance in ground effect.

I will push a change to close the hole in the preflight checks and prevent arming and flying in an alt control mode when the height validity is false.

If you wish to check the inflight fallback to loss of  GPS, then I would recommend the GPS_USE parameters.

OXINARF, 2017-10-03:
@priseborough What I'm failing to understand is why it depends on having a valid origin? If it loses the GPS after setting the origin, then the validity report will be good, even when taking off or landing. It looks like it should depend on expected takeoff/touchdown - the original use case seems to be *don't fallback to baro when ground effect happens* and not *don't fallback to baro if GPS was never good enough to set origin*.

priseborough, 2017-10-03:
We should prevent arming and takeoff if the specified altitude source unavailable. Once in flight  if GPS is lost, then Baro is better than nothing at all.

OXINARF, 2017-10-03:
@priseborough I agree, but that's not exactly what it is doing. Example, never having a GPS fix:

- takeoff in Stabilize
- gain altitude, out of ground effect
- change to alt hold

With the current code we don't do any check to enter alt hold which is clearly wrong and I agree with you that we need to check that the height is valid.

But the question now is: in this example, when the alt source is set to be GPS, should the EKF be falling back to baro and saying that it has a valid height or not? Current code will fallback but then say the altitude isn't valid so you wouldn't be able to enter alt hold (which goes against *Once in flight if GPS is lost, then Baro is better than nothing at all*).

priseborough, 2017-10-03:
Current code will fallback and report valid, because origin has been set.

OXINARF, 2017-10-03:
@priseborough My example is without ever having a GPS fix so origin isn't set.

priseborough, 2017-10-03:
That isn't a fallback

priseborough, 2017-10-03:
The vehicle should not be flying in an alt control mode in the first place in that example when we fix the checks.

OXINARF, 2017-10-04:
> That isn't a fallback

It is a fallback, it simply always fall back, not ever using GPS altitude.

> The vehicle should not be flying in an alt control mode in the first place in that example when we fix the checks.

But why? I really don't understand how *had a GPS fix, lost it, origin is set* and *never had a GPS fix, origin isn't set* are different to determine the baro fallback validity.

Also the original use case is broken: if the origin is set and GPS is lost before takeoff, then the baro altitude is valid.

priseborough, 2017-10-04:
How can you fallback from something you were never in? Anyway that's arguing over semantics and is not profitable.

If the user has specified they want to use GPS for altitude it should not be allowed to takeoff if GPS checks have never passed. I am open to options on how to achieve that.

Losing GPS before takeoff after checks have passed is a corner case compare to the very high probability of not passing checks before taking off in alt_hold.

JacksonUAS, 2017-10-17:
Any updates on which way this will progress?

roque-canales, 2018-10-08:
you do not understand.

We define EK2_ALT_SOURCE = 2.

What we ask is if the gps goes from RTK FIX to 3D Fix in flight, that EKF goes from GPS to Baro to estimate altitude.

We do not talk about helping BARO in case of no GPS.

Shrediquette, 2018-11-30:
What is the status of this 'issue' (well, with my current knowledge, I cannot tell if it is an issue or not...). 

- If GPS 3D fix is lost completely, what happens?
- If GPS goes from RTK fixed to normal GPS fix (e.g. the base station that normally sends RTK correction data is lost), what happens? Is there a way I could find information on this (without risking the integrity of my copter)?

> you do not understand.
> 
> We define EK2_ALT_SOURCE = 2.
> 
> What we ask is if the gps goes from RTK FIX to 3D Fix in flight, that EKF goes from GPS to Baro to estimate altitude.
> 
> We do not talk about helping BARO in case of no GPS.

rmackay9, 2021-08-03:
I think we can close this issue now.  Copter-4.0 (and higher) includes a check that we have a valid altitude source before arming and taking off in AltHold or any other mode that requires an altitude estimate.

I've double checked in SITL and the fallback from GPS->Baro for altitude source is working.  I tested with EKF3 after setting EK3_SRC1_POSZ = 3 (GPS) and then used SIM_GPS_DISABLE=1 to turn off the GPS.  I added some debug to display the current active source (1 = baro, 3 = GPS) and checked that the source changed to baro after the GPS was disabled.

[image]

closed 2021-08-03 by rmackay9