ArduPilot/ardupilot #7029: Two external compass are detected when using external ICM-20948

caomuqing, 2017-10-02:
#### Issue details
When using ICM20948 as external compass ON i2c bus, both compass 1 and compass 2 will be detected as external, where the COMPASS1 is detected as ICM20948, while COMPASS2 detected as AK09916.

Possible reason: the 20948 has a built-in AK09916, and its enabling bus-pass-thru.

#### Version
Arduplane 3.8.2 beta, but it should be applicable to all types of firmware

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[x] Plane
[x] Rover
[  ] Submarine

Hardware type
HERE GNSS (with ICM20948 as external compass) on a PIXHAWK2

tridge, 2017-10-02:
we need to add code to detect that we already have an AK09916 on the same bus and skip the 2nd probe

caomuqing, 2017-10-03:
The commit has been tested to work.

amilcarlucas, 2017-10-10:
Looks like it has been fixed. I'll close it now

closed 2017-10-10 by amilcarlucas