ArduPilot/ardupilot #7036: [ALL] add SR4_* streaming rate params.

sashan13, 2017-10-04:
#### Issue details
SO, my issue is that I can't properly connect AlexMos gimbal using MavLink to SR4 port. It's the only port I have free and due to me using Hacklink instead of standard FCs it's the only one I can easily connect to in my case. 

Anyways, I don't see why can't we set up streaming rates on SR4 in the same way as on other ports. Same goes for SR5, it may be needed sometimes.
#### Version
AC 3.5.3
#### Platform
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Any 

#### Hardware type
Pixhack by cuav

#### Logs
No logs.

fnoop, 2017-10-04:
This was already kind of rejected
https://github.com/ArduPilot/ardupilot/issues/2895

sashan13, 2017-10-04:
The reason why still remains unclear to me. As for request data stream
message - it is not always an option, for example when there are more then
one client on that uart.

fnoop, 2017-10-04:
@sashan13 I agree.

OXINARF, 2017-10-06:
As explained in #4303, you can only have 4 MAVLink ports configured. And USB (SERIAL0) is always one of them. Then SRx parameters are in the same order, so if SERIAL1 isn't set to MAVLink and SERIAL2 is, then SR0 applies to SERIAL0 and SR1 applies to SERIAL2 (and so on).

closed 2017-10-06 by OXINARF

fnoop, 2017-10-06:
Aaaaah.  Thanks @OXINARF for the explanation.  Very confusing, might be a better idea to tie SR4 to Serial4, SR5 to Serial5.  I've heard this same question asked many times.

OXINARF, 2017-10-06:
@fnoop I've never seen anyone confused by the other serial drivers, if you select Lidar in a serial port, then you know that the first rangefinder serial driver you selected will use that port.
I guess the confusion with MAVLink comes from confusing SR for serial, when it actually means stream rate. If it was MAVCH_x maybe people wouldn't confuse it.

sashan13, 2017-10-06:
Also, MAVLINK2 doesn't apparently count for SR parameters. This is
confusing.

Best regards,
Oleksandr Naronin

OXINARF, 2017-10-06:
@sashan13 Sure it does. MAVLink and MAVLink only says what MAVLink version you want to use, but work in the exact same way for SR parameters.

sashan13, 2017-10-06:
Well, I have all serial ports except for serial3 dedicated to MavLink.
Serial3 is for GPS, Serial0 was set to MavLink2. All serial rates are set
to non-zero and yet there was no data streamed to Serial4 until I've
changed Serial0 to old MavLink.

Best regards,
Oleksandr Naronin

fnoop, 2017-10-06:
@OXINARF - Granted I'm not the sharpest knife in the drawer, but until you just told me I assumed SR stood for serial, and I've seen this question asked lots of times by confused peeps like me.  Probably too late now, but even calling it STRx or SRTx would be clearer.

OXINARF, 2017-10-06:
@sashan13 If you have any issue please post in the forum and people will look into it.

@fnoop Yeah, I agree the name isn't the best, but it's from way back.

tscao, 2020-03-11:
Hello and sorry for digging this issue out, but I have been wanting to use the fifth mavlink for communicating with an external component and there is something I still don't understand after reading this issue and the ones related to it.

As I understand it, since 62986957b45576fbffb8d5d81c5b94c77a986f06 we have 5 possible MAVLink ports.

In the MAVLink protocol we have by default 4 ports declared (see the enum `mavlink_channel_t`) from `MAVLINK_COMM_0` to `MAVLINK_COMM_3`, but the [mavlink documentation](https://mavlink.io/en/mavgen_c/#channels) indicates that `MAVLINK_COMM_NUM_BUFFERS` can be used to change the maximum number of channels. Indeed, if I try to use a fifth MAVLink port (I opened 5 ground stations on different UDP ports), the communication is established and printing out the value of the `mavlink_channel_t chan` when sending messages gives values up to 4 (by the way I don't know if it's a good thing to use an enum value out of its bounds which go from 0 to 3 in this case).

So I don't get why we don't have a set of stream rate parameters `SR4` for a fifth MAVLink link. Am I missing something ? @OXINARF

OXINARF, 2020-03-14:
@tscao Yes, that is true, we should have SR4 since we now allow 5 ports. That change went in unnoticed for most people, but I've noticed it recently when someone else complained about it. Feel free to open a pull request with the fix!

tscao, 2020-03-18:
Okay, I am doing it, thanks for your answer !

OXINARF, 2020-03-22:
@tscao That's the number of serial ports, it's not related to the number of MAVLink links you can have - as you so well explained in your previous comment.

tscao, 2020-04-21:
You are right, `MAVLINK_COMM_NUM_BUFFERS` has not changed. I will submit a PR and link it in this message.