ArduPilot/ardupilot #7042: Rover: prevent instant response on pivot turn exit

khancyr, 2017-10-05:
#### Issue details
When doing on spot pivot turn and puting some throttle, when can pass from full backward to full forward. This should be avoid to prevent mechanical failure 

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ X] Rover
[  ] Submarine

gmorph, 2017-12-17:
Hi @khancyr.  I don't quite get what you mean.  Are you saying when you drive up to a corner in a skid steering vehicle and then it begins its turn one set of motors goes from full throttle forward to full throttle reverse and your worried about mechanical breakages?  If so there is options to slow down for corners which would mean you wouldn't be going from full forward to full reverse.  Is that sufficient?  Throttle slew speed comes to mind.  Or qre you thinking some kind of other limiting parameter here to protect the motors?

Thanks, Grant.

rmackay9, 2018-02-02:
This is fixed in Rover-3.2 (and higher).  It slows to a stop instead of instantly stopping.

closed 2018-02-02 by rmackay9