ArduPilot/ardupilot #7045: MAVLINK RC override switch

geofrancis, 2017-10-07:
i have been looking at using mavlink rc inputs via mission planner and then via a pi but the thought of only being able to regain control by clicking a button on mission planner has put me off testing it. 

Ideally if one of the auxilliary function switches could be assigned to switch between mavlink RC and regular RC  then i could just use a switch on the handset like a trainer switch where i could take control at any time from the other operator without a mad dash to click a button on mission planner.

OXINARF, 2017-10-10:
Duplicate of #1372 and #3356.

@geofrancis I'm not sure why you've opened this after commenting on one of the earlier issues...? Making repeated issues just makes the list even more useless, doesn't help at all.

closed 2017-10-10 by OXINARF

geofrancis, 2017-10-10:
#3356 was about simultaneous control, and i tried searching for mavlink rc but I didnt find #1372, its name is not easy to find.

lebarsfa, 2021-05-14:
Hello, 

Please correct me if I am wrong, if I understand well this functionality is now possible as indicated in https://github.com/ArduPilot/ardupilot/blob/7cf4195f86efb9bc260feb5e85a0c0fdde6c6e52/ArduCopter/ReleaseNotes.txt#L830 e.g. by setting RC10_OPTION to 46. 
This is probably related to #8938 and probably solves #1372, and partially #3356.

I am adding this comment because while reading those different issues (and others mentioned inside), it was not clear to me how to get the functionality described in this issue (I have not yet tested though, I will try to do it soon...).

Thanks.

peterbarker, 2021-05-17:
@lebarsfa yes, you can now use an RC_Channel option to disable RC-channel ("mavlink") overrides.