ArduPilot/ardupilot #7050: Disable use of guided waypoints until after the vehicle has taken-off

nick2204, 2017-10-09:
#### Issue details
Improvement for specific Quadplane configurations
Disable use of guided waypoints until after the vehicle has taken-off - prevent sudden movement as vehicle leaves the ground

Possibly combined in a config word with various other vehicle specific settings.

#### Version
Arduplane 3.8.2-dev (running in SITL)
Q_enable =1 (quadplane)
Q_guided_mode = 0 (FW loiter)
Q_RTL_mode = 1 (land in VTOL once RTL reached)

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Quadplane

#### Hardware type
PH2.1
SITL

nick2204, 2017-10-13:
This was found as a result of the strange behaviour in QGroundcontrol. When start mission dialog box is used it causes a switch to guided mode before the mission is started.
Setting mode to auto and manually  arming appears to be ok.
QGC issue to be raised.

tridge, 2017-11-13:
I think a min height before navigation makes sense here, but we should also get to the root of the QGC behaviour to understand the ordering of operations