ArduPilot/ardupilot #7053: Plane: Land Abort should level off briefly before pitching up

magicrub, 2017-10-09:
Current behavior on all LAND ABORT triggers:
- instantly throttle up and pitch up

Problem:
- throttle may take a moment to take effect so pitching up may cause a stall since in many cases we are performing a slow approach airspeed

Desired behavior:
- instantly throttle up
- level pitch and hold briefly (2 seconds?)
- continue pitching up like normal.

WickedShell, 2017-10-09:
I'd disagree with that, if you ever stall then something is off with TECS/airspeed tuning, land abort should never be entering an unachievable range on pitch that is outside the requested airspeed envelope. Trying to hold level for a minute is going to increase the chance of hitting something, we should always be doing the best climb we possibly can immediately.

magicrub, 2017-10-09:
after discussing with @WickedShell on mumble, this is generally a bad idea. An externally triggered abort should immediately react. Closing..

closed 2017-10-09 by magicrub