ArduPilot/ardupilot #7055: Plane: Implement DO_LAND_CFG

magicrub, 2017-10-09:
proposal by @WickedShell and I have to make a new command DO_LAND_CFG  that is to be called directly after a DO_LAND_START. Ideas of what this could configure:
- generate new approach heading via wind with various possible restrictions
- change airspeeds depending on wind conditions
- enable/disable fly-over land pt before approach (to detect baro drift via rangefinder)
- ????

geofrancis, 2017-10-10:
This sounds good, auto landing works ok when there is very little wind but when the wind can change direction 90 degrees between take off and mission end so I find landing in FBWA more reliable.

airspeed sensor logic needs attention. I found that when flying a large draggy airframe in high wind the airspeed sensor would cause it to stall especially on landing. when a gust hit the plane it read that as high airspeed throttled back and due to the high drag airframe it lost almost all its ground speed in a very short time, this lead to a stall as soon as the gust of wind subsided that could not be recovered from at such a low altitude. almost like a small scale microburst  I found this was not an issue when the airspeed was disabled, so I currently have it set to reject airspeed data if it diverges from ground speed by too much, so it should be disabled if wind speeds get too high.

WickedShell, 2017-10-12:
My initial proposal for the MAVLink side is on the `wickedshell/land-cfg-proposal` branch [here](https://github.com/WickedShell/mavlink/tree/wickedshell/land-cfg-proposal).

What it outlines is the start to a system where multiple parameters for a landing can be outlined after a DO_LAND_START command. This is done by with mission items of `MAV_CMD_DO_LAND_CFG` where param1 indicates what sub type of landing configuration this is. The examples outlined so far all have to do with aligning the flight path with the wind, or avoiding certain angles. For example a `MAV_LAND_CFG_INTO_WIND_DEG` would tell the aircraft to rotate the landing up to so many degrees to be perfectly into the wind. If the user wanted to it could then be followed with a series of `MAV_LAND_CFG_KEEPOUT_ARC` items to indicate what angles the aircraft may not fly a landing through. (This could be used to avoid overflight of an area, or to ensure the aircraft is aligned with a runway).

Internally the autopilot would process the landing configuration and fly the resultant path. If no path can be arrived at due to excessive restrictions (IE all headings are marked keepout, or the aircraft can't rotate the path far enough to avoid a keepout zone) the aircraft would refuse to land and would execute an RTL. If in a failsafe condition we should simply fly the planned landing pattern as it's better then falling out of the air.

Because slope and deepstall landings interpret the outcome of a rotation command differently it would be up to each landing backend to determine what to do with the output of the configuration, but the actual landing configuration processor would be common to all backends. Internally there will have to be a limit to the number of `MAV_LAND_CFG` commands the autopilot can process due to memory and processing complaints, but it should be quite feasible to allow at least 5 commands to be processed.

Ideally this system should allow us to specify a number of different additional configuration parameters in the future, but the main use case I had in mind and have thought through thus far is related to wind directions and the handling of them.

For this to be less painful for the user I imagine a GCS will play a large role in hiding the internal use of the various `MAV_LAND_CFG` values and will present a single landing item where it generates the appropriate commands when uploading the mission much the way GCS's can hide the commands used for a survey mission. @DonLakeFlyer @meee1 do you for see any problems with this from a GCS side?

pompecukor, 2017-11-08:
Please please please:
Deploy landing gear please (happy with just a simple servo set command or override servo command. Pity it is limited to 1-8 ch, but we take what we get). This as you can guess can be useful for multiple stuff, like close camera cover or close belly bay for landing protections and so on.

Naterater, 2018-12-01:
Still interested in this moving forward.

pompecukor, 2018-12-01:
Definitely Please. This would be killer feature. especially flyover to detect drift. Would make all landing 100% precise :)

magicrub, 2018-12-09:
I have a feeling this will be better implemented using scripts.