ArduPilot/ardupilot #7058: Error when calibrating if secondary compass disabled

fredrikfalkman, 2017-10-10:
I have never (I think) been able to get both compasses calibrated, but with secondary compass disabled (COMPASS_USE2 = 0) things have worked smoothly until I upgraded to 3.8.1
Now if I disable the secondary compass I get "MAV_CMD(42424) command failed" and the calibration process halts.
Is this a known error? Suggestions? Thanks!

#### Version 3.8.1 (QGC 3.2.1)

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Hardware type
Navio2

OXINARF, 2017-10-11:
@fredrikfalkman I would say this is a QGC issue, but let's see what Don has to say about it.

@DonLakeFlyer Why is QGC sending specific mag bits (https://github.com/mavlink/qgroundcontrol/blob/master/src/AutoPilotPlugins/APM/APMSensorsComponentController.cc#L264) instead of using the all value (0)? The reason why @fredrikfalkman is getting a failure is that QGC is asking to calibrate a compass that the user disabled, so ArduPilot purposely fail the command.

DonLakeFlyer, 2017-10-21:
I can't quite remember but I think it had something to do with being able to get individual status back for each compass being calibrated. That way I can show progress for each compass. Let me run that code again to see. I think it used to be ok to send bits for compasses which were disabled. Looks like that changed?

DonLakeFlyer, 2017-10-21:
Fixed: https://github.com/mavlink/qgroundcontrol/pull/5772

fredrikfalkman, 2017-10-21:
Nice!

OXINARF, 2017-10-22:
@DonLakeFlyer 

> I can't quite remember but I think it had something to do with being able to get individual status back for each compass being calibrated.

I'm pretty sure that works even when sending the *calibrate all* command.

> I think it used to be ok to send bits for compasses which were disabled. Looks like that changed?

There was a change but it was one year ago... If you haven't read it, then I recommend looking at https://github.com/ArduPilot/ardupilot/pull/5061 where this change (and others) were made - namely reports will always be sent until GCS acks it has received them (the board has to be restarted anyway).
I wasn't aware that QGC actually used the mask instead of the _calibrate all_ option or I would have notified you (I actually opposed the change, but lost 🙂).

Glad it's fixed now, thanks!

closed 2017-10-22 by OXINARF