ArduPilot/ardupilot #7061: Plane add CRASH_DETECT=2

magicrub, 2017-10-10:
When using TKOFF_THR_MINACC >0 for hand launches, the current crash detection (=1) will immediately disable the motor thinking it was a failed bungee/catapult launch. Adding an option of (=2) would offer a longer delay (10s?) so that you can shake to enable the motor and still have plenty of time to hand launch before it thinks something is wrong and disarms.

sashan13, 2017-11-27:
I believe you're doing it wrong. I mean shaking. Correct way of doing it would be just tossing it with motor disabled - it registers acceleration when being tossed and enables motor immediately.
Still, some other variants of behaviour may be good.

magicrub, 2017-11-27:
@sashan13 some aircraft need power immediately or else don't survive the hand launch. The prop spin-up delay after the throw is too much. Light aircraft are easy to launch, heavy ones not so much.

kikislater, 2018-04-28:
Hi, 
Sorry for previous bad PR. 
This feature is needed in case of shaking users needed to disable crash detection which could be a safety lack.

Naterater, 2019-01-05:
Just want to link these two together to keep things relevant: https://github.com/ArduPilot/ardupilot/issues/2221