ArduPilot/ardupilot #7062: Failure to transition, and uncontrolled climbing due to interaction of VTOL & FW profiles

nick2204, 2017-10-10:
#### Issue details
In certain conditions it is possible for a VTOL aircraft to fail to complete transition and continue climbing due to an interaction of the VTOL and FW power systems.

During transition If the current altitude is higher than the desired altitude, the aircraft attempts to lower the forward thrust to in order to descend, as the pitch control is restricted in this state, but if this prevents the vehicle reaching transition speed, then the VTOL motors can prevent it from loosing altitude.

This can lead to a situation where transition speed is never reached, and Altitude continues to climb.

#### Version
Arduplane 3.8.2-dev (running in SITL)
Q_enable =1 (quadplane)
Q_guided_mode = 0 (FW loiter)
Q_RTL_mode = 1 (land in VTOL once RTL reached)

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Quadplane

#### Hardware type
PH2.1
SITL
#### Logs
[SITLROM_PARAM.zip](https://github.com/ArduPilot/ardupilot/files/1373622/SITLROM_PARAM.zip)

nick2204, 2017-10-10:
Generally this state can be avoided by ensuring that enough distance/time is left left between transition and FW loiter.

Fix could simply be prevention of FW loiter until transition has completed. 
Alternately if Loiter is required vehicle could either:

- VTOL loiter
- finish transition on current heading then return to location for FW loiter

tridge, 2017-10-21:
I've fixed this properly with PR #7102

OXINARF, 2017-10-22:
#7102 is now merged.

closed 2017-10-22 by OXINARF