ArduPilot/ardupilot #7065: Bug in AP_Notify/Display, which accidentally doesn't harm

olliw42, 2017-10-11:
Display_SSD1306_I2C.h:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Notify/Display_SSD1306_I2C.h#L9

SSD1306_ROWS_PER_PAGE is defined, and the name implies it should be the number of rows per page, and not the number of pages, but it is actually used for both numbers. Coincidentally, for the the specific 128x64 display, the number of pages and the number of rows per page are identical (equal to 8), but that's not neccessarily so for other display formats. This mixup could thus lead to potential memory issues, if in later stages other displays are added.

For instance:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Notify/Display_SSD1306_I2C.h#L34
The size of _displaybuffer is determined as [SSD1306_COLUMNS * SSD1306_ROWS_PER_PAGE], however it should be SSD1306_COLUMNS * (number of pages)

One way to resolve that could be to have two defines

#define SSD1306_PAGES  8
#define SSD1306_ROWS_PER_PAGE (SSD1306_ROWS / SSD1306_PAGES)

or vice versa, depending on what is liked better, and to change SSD1306_COLUMNS * SSD1306_ROWS_PER_PAGE to SSD1306_COLUMNS * SSD1306_PAGES, wherever it is required, that is here:

* https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Notify/Display_SSD1306_I2C.h#L34
* https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Notify/Display_SSD1306_I2C.cpp#L76
* https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Notify/Display_SSD1306_I2C.cpp#L154

and to change (SSD1306_ROWS / SSD1306_ROWS_PER_PAGE) to SSD1306_PAGES, wherever it is required, that is here:

* https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Notify/Display_SSD1306_I2C.cpp#L120

The same comments apply to the files Display_SH1106_I2C.h, Display_SH1106_I2C.cpp.

lvale, 2017-10-30:
@olliw42 Just got a SH1106.

Would you mind that I take an attempt of implementing your suggestions and test and submit a PR ?

olliw42, 2017-11-03:
no I don't mind at all, why should I, in the contrary, I posted the finding so that it is known :)
thx, sir

olliw42, 2023-02-03:
closing it since no one cares :)

closed 2023-02-03 by olliw42