ArduPilot/ardupilot #7066: QuadPlane: Q_YAW_RATE_MAX doesn't take effect

expntly, 2017-10-11:
#### Issue details

It seems changing Q_YAW_RATE_MAX doesn't actually change the yaw rate.  Here's an AUTO mission with a 90-degree turn.  The flight at the top is with Q_YAW_RATE_MAX=20, the one at the bottom is with Q_YAW_RATE_MAX=10.

[image]

#### Version
ArduPlane-3.8.1 SITL

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
QuadPlane

#### Hardware type
Happening with Pixhawk2.1, but reproduced with SITL.

nick2204, 2017-10-13:
Are you sure that the this is the limiting factor? 
How fast is it yawing? never seen a quad plane be able yaw too fast except when it yaws one way much faster than the other from mechanical alignment issues.

tridge, 2017-10-19:
Q_YAW_RATE_MAX is a scaling factor for pilot input on rudder stick. It doesn't affect AUTO missions.
I'll update the docs

tridge, 2017-10-21:
docs have been updated, closing

closed 2017-10-21 by tridge