ArduPilot/ardupilot #7067: Can't set RNGFND_LANDING for Copter

sbuebel, 2017-10-11:
#### Issue details
Trying to get Lidar Lite V2 to work with a quadcopter running APM 3.5.3 and using Mission Planner 1.3.49 (don't think this really matters). It is connected via PWM and is not working, just giving me zeros and saying bad LIDAR health. I have researched this and found that on the plane platform, there is a parameter "RNGFND_LANDING" that needs to be set to 1 for this to work. Is there an equivalent for the copter? Has anyone experienced this issue?

#### Version
APM V3.5.3
Lidar Lite V2
Mission Planner 1.3.49

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Quadctoper

#### Hardware type
Pixhawk

magicrub, 2017-10-11:
You've asked a "I can't get this working, please help" question on the bug reporting and feature request page. Please re-post on the forums at discuss.arudpilot.org so people can give you assistance with where it is on the wiki.
Closing

closed 2017-10-11 by magicrub

sbuebel, 2017-10-11:
I was thinking this issue is a bug because the parameter doesn't exist in Copter 3.5 but is necessary for LidarLite PWM implementation.

magicrub, 2017-10-11:
Please re-post on the forums at discuss.arudpilot.org so people can give you assistance with where it is on the wiki.