ArduPilot/ardupilot #7069: Dead when armed:Assertion failed at file:armv7-m/up_hardfault.c line: 184 task: ArduPlane

ranqingfa, 2017-10-12:
#### Issue details
Serial 5 console output:
[init] looking for microSD...
[init] card mounted at /fs/microsd
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
[init] USB interface connected
Running rc.APM
Mounting binfs
binfs mounted OK
nsh: rm: unlink failed: No such file or directory
uorb started OK
Starting ArduPilot
Starting ArduPlane uartA=/dev/ttyACM0 uartC=/dev/ttyS1 uartD=/dev/ttyS2 uartE=/dev/ttyS6
initialised /dev/ttyS3 OK 1024 5ArduPilot started OK
12
rc.APM finished
initialised /dev/ttyS1 OK 1024 5
12
NuttShell (NSH)
initialised /dev/ttyS2 OK 1024 5nsh> 12
initialised /dev/ttyS6 OK 1024 512
Starting driver mtd start /fs/mtd
mtd: started OK
Starting driver mtd readtest /fs/mtd
reading /fs/mtd expecting 16384 bytes
readtest OK
mtd: readtest OK
Failed to load defaults from /etc/defaults.parm
Starting driver adc start
adc init done
ADC started OK
Starting driver px4io start norc
px4io default PWM output device
px4io started OK
Starting driver px4io checkcrc /etc/px4io/px4io.bin
px4io: CRCs match
PX4IO CRC OK
Starting driver fmu mode_pwm4
fmu mode_pwm4 started OK
SPI device mpu6000_ext on 4:1 at speed 500000 mode 3
mpu6000_ext: reg 75 68 ff
SPI device mpu6000_ext on 4:1 closed
SPI device mpu9250_ext on 4:1 at speed 1000000 mode 3
mpu9250_ext: reg 75 71 ff
SPI device mpu9250_ext on 4:1 closed
SPI device icm20608_ext on 4:1 at speed 1000000 mode 3
icm20608_ext: reg 75 af ff
SPI device icm20608_ext on 4:1 closed
SPI device icm20608_ext on 4:1 at speed 1000000 mode 3
icm20608_ext: reg 75 12 ff
SPI device icm20608_ext on 4:1 closed
SPI device icm20608-am on 1:2 at speed 500000 mode 3
icm20608-am: reg 75 af 00
SPI device icm20608-am on 1:2 closed
SPI device icm20608-am on 1:2 at speed 500000 mode 3
icm20608-am: reg 75 12 00
SPI device icm20608-am on 1:2 closed
SPI device lsm9ds0_am on 1:2 at speed 11000000 mode 3
lsm9ds0_am: reg 0f 49 49
SPI device lsm9ds0_am on 1:2 closed
SPI device mpu6000 on 1:4 at speed 500000 mode 3
mpu6000: reg 75 68 68
SPI device mpu6000 on 1:4 closed
AP_I2C_0 on I2C bus 1 at 0x55 (bus: 100 KHz, max: 100 KHz)
SPI device ms5611 on 1:3 at speed 20000000 mode 3
MS5611 found on bus 1 address 0x03
AP_I2C_1 on I2C bus 1 at 0x28 (bus: 100 KHz, max: 100 KHz)
I2C device bus 1 address 0x28 closed
AP_I2C_2 on I2C bus 2 at 0x28 (bus: 100 KHz, max: 100 KHz)
MS4525: no sensor found
I2C device bus 0 address 0x28 closed
AP_I2C_3 on I2C bus 1 at 0x1e (bus: 100 KHz, max: 100 KHz)
I2C device bus 1 address 0x1e closed
AP_I2C_4 on I2C bus 2 at 0x1e (bus: 100 KHz, max: 100 KHz)
I2C device bus 0 address 0x1e closed
SPI device hmc5843 on 1:5 at speed 11000000 mode 3
SPI device hmc5843 on 1:5 closed
SPI device lsm9ds0_am on 1:2 at speed 11000000 mode 3
Failed to load defaults from /etc/defaults.parm
SPI device mpu6000 on 1:4 at speed 500000 mode 3
SPI device lsm9ds0_g on 1:1 at speed 11000000 mode 3
SPI device lsm9ds0_am on 1:2 at speed 11000000 mode 3
disabling flow control on /dev/ttyS1 _total_written=346
disabling flow control on /dev/ttyS2 _total_written=70
Assertion failed at file:armv7-m/up_hardfault.c line: 184 task: ArduPlane
sp:     2000e4b8
IRQ stack:
  base: 2000e4fc
  size: 000002e8
2000e4a0: 080e528c 000000b8 080e51c6 2000e4b8 1000aea8 080b6eb5 080e528c 000000b8
2000e4c0: 080b0391 00000010 2000ca00 00000003 00000000 080b7145 080b7131 080bfcd9
2000e4e0: 00000000 00000000 1000ad9c 000034a8 ecb3b411 080b70b5 1000aca4 00000000
sp:     1000ad78
User stack:
  base: 1000aea8
  size: 00001ffc
1000ad60: bff5f35c 42c80000 00001387 0000000c 20000010 080b4001 20002a10 00000013
1000ad80: 20000b50 20000b50 00000004 00000012 00000010 0809c231 200051a0 00000000
1000ada0: 00000001 0809c229 20002a10 00000013 20000b50 08038509 20002a10 00000000
1000adc0: 00000001 00000000 20002a10 0800a767 20002a10 00000004 00000000 0800a659
1000ade0: 20002a10 00000004 00000001 2000002c 00000168 08038d9f 00000000 20002a10
1000ae00: 20000b50 08009783 080096d1 20002918 00000001 08039985 20001050 00000012
1000ae20: 00004bb4 08045203 20001050 00004e20 00003728 00000001 200092c0 20009310
1000ae40: 20009844 00000000 20016ce0 20016c20 20009738 0809918f 080991e9 00000000
1000ae60: 20012338 00000000 00000000 00000000 00000000 00000000 080991e9 00000000
1000ae80: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 080b171b
1000aea0: 00000000 00000000 ffffffff 62e4bd1e 0ac550d6 840ceaa7 00000410 80002010
R0: 00000000 2002fff8 ecb395a3 0cb4d3b9 00000000 1000ad9c 000034a8 ecb3b411
R8: ecb395a3 00000003 2000c87c 00000000 1000ada8 1000ad78 080b0f51 080b400c
xPSR: 81000000 BASEPRI: 00000000 CONTROL: 00000000
EXC_RETURN: ffffffe9

#### Version
3.8.0/3.8.2

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
fixedwing

#### Hardware type
pixhawk

#### Logs

tridge, 2017-10-19:
we need to get your parameters - can you give a complete parameter file?

ranqingfa, 2017-10-21:
Here is the parameter list I am using
http://pan.baidu.com/s/1mhTqjf6

ranqingfa, 2017-10-27:
Any answer?
https://drive.google.com/file/d/0B6nj4A2GqXmWT0tRdHVheEtXYUE/view?usp=sharing

tridge, 2017-11-12:
That parameter file isn't from a 3.8.x release, as it includes ELEVON_OUTPUT, which was not in 3.8.0
It is possible the bug still exists though, and is likely to be quite tricky to track down.
Can you first of all try to reproduce this on 3.8.3 (or beta if not released yet). Then if you can reproduce, try disabling non-default features until the problem disappears. That may help us track it down.
Also, please include a tlog from a GCS when the crash occurs. That will give us some idea of the timing relative to what is happening in the code.

RQRobert, 2019-03-27:
Hi There
I have seen you run the arducopter code in nutshell.

I want to know how to startup arducopter code from Nutshell Prompt.

peterbarker, 2021-07-30:
No response from OP in years to questions that need answering; closing.

closed 2021-07-30 by peterbarker