ArduPilot/ardupilot #7071: Plane: missin a simple relay control

robustini, 2017-10-12:
#### Issue details
I need to control a simply relay circuit  connected to the Pixhawk "AUX5" pin, I also use it with Raspberry and it works fine, usefull for control lights and other thing, this one:

[image]

I want to use a free tx channel, like 13, for control the relay state (on/off)
In the "SERVO_FUNCTION_13" it would be useful to have a "Relay" command enabled by tx switch, but I don't see this assignment in the "Plane", but it is in Copter.

#### Version
Plane 3.8.2

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
It's not important

#### Hardware type
Pixhawk

#### Logs
It's not important

khancyr, 2017-10-12:
It is also valid for rover. We need more option on channel and new switch functions!

geofrancis, 2017-10-12:
i was going to use this function on a plane but luckily i read this and seen it didnt work before rewiring it!

robustini, 2017-10-12:
There's a lot of function in the Copter also useful in Plane, like "Parachute", "Landing gear (useful for some VTOL)" and other.
It's a shame that there are not even in the Plane.

kabturek, 2017-10-12:
so this arduplane wiki is wrong ? http://ardupilot.org/plane/docs/common-relay.html 

Wanted to use it on my new build for switching video source (the slow boat video switch is on the way) today.

khancyr, 2017-10-12:
it works with do_set_relay mavlink command but only copter got the Ch7/Ch8 switches

robustini, 2017-10-12:
The wiki is related to the Copter.

docs/common-relay.html

kabturek, 2017-10-12:
@robustini  from what i understand the chapter "Pilot control of the relay" is related only to the copter - the rest applies to plane (why would it be on the plane wiki if not?) 
IMO that chapter about "Pilot control of the relay" should be deleted to not cause confusion

geofrancis, 2017-10-12:
Aux functions from copter that would be usefull:

4 | RTL
5 | Save Trim
7 | Save WP
9 | Camera Trigger
10 | RangeFinder
11 | Fence
14 | Acro Trainer
15 | Sprayer
16 | Auto
17 | AutoTune
21 | Parachute Enable
22 | Parachute Release
23 | Parachute 3pos
24 | Auto Mission Reset
27 | Retract Mount
28 | Relay On/Off
34 | Relay2 On/Off
35 | Relay3 On/Off
36 | Relay4 On/Off
30 | Lost Copter Sound
37 | Throw
38 | ADSB-Avoidance
39 | PrecLoiter (for VTOL)
41 | ArmDisarm

I always thought that this should be in the RC INPUT function parameters rather than in a separate RC FUNCTION parameter.

robustini, 2017-10-12:
@kabturek, i agree!

magicrub, 2017-10-12:
Now that VTOL is a copter-like feature that lives in Plane, it's a good idea to merge more features together. this is one of those features. AUX functions that are in copter are treated totally differently in plane and they're not even called AUX and can not be configured the same way. Some good to add to Plane/Rover/Sub

peterbarker, 2017-10-12:
On Thu, 12 Oct 2017, Tom Pittenger wrote:

> Now that VTOL is a copter-like feature that lives in Plane, it's a good idea
> to merge more features together. this is one of those features. AUX
> functions that are in copter are treated totally [...]

That's where we're hoping to go with 
https://github.com/ArduPilot/ardupilot/pull/6819

robustini, 2017-10-13:
@peterbarker, i did not see your implementations or I would not open this issue, i'm sorry.
Yeah, this is the right direction, when you think commit it in the master?

kabturek, 2017-10-13:
awesome stuff @peterbarker ! how can we help ? ;-)

peterbarker, 2017-10-14:
@robustini @kabturek That PR is currently blocked by another PR, https://github.com/ArduPilot/ardupilot/pull/6998 - that's currently blocked on some externalities, as @OXINARF has noted that one GCS will not take kindly to the change in the markup in the code and another consumes an `xml` file it really shouldn't...

@geofrancis Glad you noticed those similarities.  `RCMAP` seems to be a natural thing to move in - but someone did note that it's going to be a serious disruption for every GCS out there!

In terms of helping
 - testing the PR wouldn't go astray!  It's a good idea to use a vehicle you're quite happy to nuke all parameters on, however....
 - helping with the externalities (if they're still considered blockers) would be helpful.
 - Reviewing the PR
 - helping me work out whether removing the `AP_PARAM_FRAME_COPTER` when we move to having it on Rover as well will have things work as intended....

robustini, 2017-10-25:
Thanks to everyone, so to have those features we should wait for the version 5.0 of the Plane? :-)
What I ask at the moment is the ability to set in "SERVOx_FUNCTION" a function for contror a digital output, not only PWM.

gmorph, 2017-12-17:
I've merged in #6998 - working on #6819 but its a big change so will take a little time.

commit 901664ce (2018-09-05): Plane: add support for RCn_OPTION (servorelay,camera)

Closes #7071
Closes #7666
closed 2018-09-05 by peterbarker