ArduPilot/ardupilot #7072: New parameter for VTOL plane safe transition distance on RTL

nick2204, 2017-10-13:
#### Issue details
Currently VTOL planes use WP_LOITER_RAD as a boundary to switch from RTL to QRTL (when Q_RTL_MODE is set) as it approaches home.

As WP_loiter radius has other uses, particularly relating to guided mode FW flight, it makes sense to define a safe transition distance for a particular airframe and leave it set.

example: Failsafe RTL is triggered while doing a very wide loiter (600m) aircraft will try to fly in VTOL mode from 600m from home location.

#### Version
Arduplane 3.8.2-dev (running in SITL)
Q_enable =1 (quadplane)
Q_guided_mode = 0 (FW loiter)
Q_RTL_mode = 1 (land in VTOL once RTL reached)

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Quadplane

#### Hardware type
PH2.1
SITL
#### Logs

robustini, 2017-10-17:
I agree.
There's also another problem to be solved, always tied to this.
In case of involuntary QRTL, for example during a battery failsafe procedure, if the VTOL is at a considerable height (for example 200 meters) and close enough to the home performs the FW-> MC transition immediately, and will have to make a vertical descent in copter putting a long time before getting to the ground and land.
So it would be useful first the Plane make a Loiter descent in fixed wing up to a preset height (for example 20 meter), and then make the transition.
Battery consumption would be significantly lower.

nick2204, 2017-10-17:
Good point. I think in most of my scenarios, a wider descending loiter to a predefined height prior to transiting would work nicely.

nick2204, 2017-10-23:
@tridge how do you feel about this approach for RTL?
- Fly back to "Q_RTL_RADIUS" in FW (proposed new param)
- If altitude > "Q_RTL_ALT" then remain in FW and do spiral descent
- once "Q_RTL_RADIUS" & "Q_RTL_ALT" are met transition towards home and complete QRTL

tridge, 2017-11-12:
I've fixed the distance from home for QRTL transition for 3.8.3, adding a new Q_TRANS_DECEL parameter. 
I haven't added spiral descent yet

tridge, 2017-11-12:
I've changed the subject for the remaining issue

tridge, 2017-11-12:
another approach would be to use reverse thrust in quadplane transitions, so we can slow down without gaining height

nick2204, 2017-11-15:
Yes, reverse thrust could be quite good for improving stability and efficiency in transition, but does not solve the issue of QRTL being very long if the vehicle crosses the threshold of the QRTL radius before it has descended, particularly in a fail-safe triggered RTL scenario.

Also i have now tested the dynamic QRTL radius with Q_TRANS_DECEL and it works very nicely once configured correctly.

DavidIngraham, 2018-01-25:
@nick2204 I believe it is possible to get the behavior you are looking for using mission items that are currently available. This requires some effort before the flight, and I agree that it would be nice to have this built into QRTL, but the following is currently possible:

- Configure RTL_AUTOLAND=2. This will make the vehicle jump to the closest DO_LAND_START item in the mission when RTL is initialized
- Create a landing plan with the following items:

1. DO_LAND_START
2. Waypoint Near the location where you'd like to loiter down (Optional, use this if you want to prevent the vehicle from descending until it has reached the loiter point. This could also be omitted if you have RTL_AUTOLAND=1)
3. LOITER_TO_ALT, with your target altitude for transition
4. Waypoint (optional, include if you want to be explicit about where the transition begins after loiter)
5. VTOL_LAND, at the landing point

This would be even better if we get https://github.com/ArduPilot/ardupilot/issues/7509 fixed because it would allow us to use Q_TRANS_DECCEL to determine the transition point in an auto landing plan.

Another advantage of this is that it lets you specify the heading of the aircraft during transition. This is helpful if there is any wind or if you are in a location with trees, powerlines, or other hazards that may be in the transition path. With a QRTL, the heading during transition isdependent on where the RTL is initialized.

DavidIngraham, 2018-01-25:
[image]
Here is an example mission that I just flew in SITL. The vehicle takes off, then flies in a box patter at 400 ft until RTL I call RTL.

Upon RTL, the vehicle proceeds to point 9 at an alt of 400 ft, then loiters down to 75ft. It then flies as a fixed wing to point 11, transitions, and lands as a VTOL.

magicrub, 2018-01-25:
those are in meters, not feet.

DavidIngraham, 2018-01-25:
I have the units set to feet in mission planner. It automatically converts the altitudes on the flight data page. 

Sorry for the confusion, It would have been better to use meters. I'm a dumb american. :grinning:

magicrub, 2018-01-26:
>  I'm a dumb american

https://pics.me.me/burrito-justice-burritojustice-quick-while-the-government-is-shut-down-30477316.png

nick2204, 2018-02-07:
@DavidIngraham, Thanks, definitely possible to program that behavior manually in a mission or simply manual override if necessary.

The key to this was have some consistency in the RTL/transition behaviour from the power planning POV.
If the aircraft is far away, it will already try to descend to the RTL/QRTL altitudes before transition.
But the unexpected triggering of RTL too close to home doesn't allow the time to descend before transition.
Generally the mission will not have any intention of an RTL clsoe to home, so to add a whole lot of extra landing details may become tedious

In the end you know you never want to hover vertically from say 120m to ground level, it just consumes too much power, so why not determine a safe default that can be planned for in you power calculations.

pompecukor, 2018-07-24:
This would be so awesome if if can be added. The only current "safe" option is to add lot of line in the mission, just as David did, which is so counter intuitive. I wish I could code :)

DavidIngraham, 2018-11-13:
This functionality was added in https://github.com/ArduPilot/ardupilot/pull/9707

closed 2018-11-13 by DavidIngraham