ArduPilot/ardupilot #7075: VTOL: in AUTO mode turn the plane towards the first waypoint after the transition MC->FW, not during

robustini, 2017-10-15:
#### Issue details
During the transition from MC to FW in AUTO, if the first WP is in the opposite direction of the nose
the VTOL performs the turn during the transition (see the image).
It would be desirable instead to perform the turn towards the first waypoint just after the transition, not during, when she completed it and went to FW.

[image]

#### Version
Plane V3.8.2

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

magicrub, 2017-10-16:
Wouldn't it make more sense to yaw while in VTOL mode towards the next waypoint? There would need to be an altitude or height check after liftoff because we don't want yawing a few cm off the ground!

robustini, 2017-10-16:
Maybe I explained myself wrong.
I mean the correct procedure is this:

1) takeoff in Copter mode
2) when the takeoff height has been reached transition in Plane mode
3) when the transition in Plane mode are complete yawing to the first waypoint

magicrub, 2017-10-16:
I'm wondering why you would want to start flying forward in the wrong direction at all? Why not yaw around into the correct heading while still gaining altitude but before the transiton.

robustini, 2017-10-16:
Because we can not think that any VTOL user is scrupulous... ;-)
Also the first waypoint could have been scheduled automatically and not necessarily in the direction of takeoff (upwind).

robustini, 2017-10-16:
A classic example is the normal Plane: during the automatic takeoff the Plane proceeds in the nose direction until it reaches the predetermined height, then performs in the case of turning to the first waypoint, not during the takeoff.

magicrub, 2017-10-16:
because "into wind". Got it.

nick2204, 2017-10-16:
I agree, more control of what happens during transition would be good, it is really important to get the transition done as quickly as possible. The issue i submitted ( #7062 ) is another example of what happens if transition is not allowed to finish before navigation commences.

So far my workaround is always using weather vanning in VTOL, and then always putting a waypoint upwind at a safe distance for transition. Worst case distance is generally no wind, seeing that is always heading to wind.

It could be possible to make a combined solution for these and the request i made ( #7072 )for an independent safe transition distance on RTL

tridge, 2017-10-21:
I'd be happy to add this as an option, but not to always wait for the turn. The reason is that where I fly if the plane goes straight ahead it could fly out of bounds if it doesn't turn (eg. fly over the clubhouse, or over the pilot). Waiting till transition is completed would be very bad for my flying location.
I would be happy to add a parameter to control this though. For example, we could have a maximum bank angle during transition, and/or have a flag for whether multicopter yaw control is applied during transition.
Perhaps we need a set of flags? maybe Q_OPTIONS, and have a bitmask of flags of this type? Users could set the bit to allow/disallow turns during transition.

robustini, 2017-10-21:
Hello @tridge , thanks for the answer.
You know how VTOL I'm flying, in some configurations where the copter section has not been adequately sized the flight can become critical during that transitions, so it would be preferable to keep the wings leveled until the end of the transition, that is until the wing profiles begin to work to support the flight.
Usually to avoid this behavior before takeoff on an automatic mission I physically put the VTOL bow already directed to the position of the first waypoint, but in the case of automatic planning not necessarily the ground station software can predict my takeoff direction, I would love that there was this option only for that reason.
The max bank angle could be a good solution, in this way you can set the maximum deviation, if set to 0, it must be leveled.
I don't think we can use a bitmaks with an existing "Q" parameter for this feature, I've checked all.

tridge, 2017-10-21:
yes, if all we want is a roll limit, then we should have a Q_TRANS_ROLL_MAX parameter. That would allow it to yaw using the multirotor control, but keep wings level.
Any other suggestions for a name for the parameter? It is certainly very easy to add.

robustini, 2017-10-21:
Q_TRANS_ROLL_MAX, why not? Perfect!

tridge, 2017-10-31:
This is fixed in 3.8.3beta1 using the new Q_OPTIONS flags. If you set Q_OPTIONS to 1 then that enables the level transition code. The wings are held within LEVEL_ROLL_LIMIT degrees until the transition is complete.
I re-used the existing LEVEL_ROLL_LIMIT parameter rather than adding a new parameter

closed 2017-10-31 by tridge