ArduPilot/ardupilot #7081: broken support for MTK GPSes

night-ghost, 2017-10-17:
this commit (https://github.com/ArduPilot/ardupilot/commit/2f8f2ffd2dec1d85546cc9b8f2e875d9b77fb4c9) removed initialization for NMEA GPS. But binary protocols for MTK is under "#if !HAL_MINIMIZE_FEATURES" so don't present in precompiled binaries. In result, GPS based on MTK chipsets, even on world's best MT3333, now don't works correctly without recompiling or tuning GPS via special tool.

WickedShell, 2017-10-17:
Shoot, sorry I missed that MTK & SIRF are both inside the minimize feature set, that was an unintentional breakage. They were removed on the philosophy that all of those drivers had a proprietary protocol that was better then the NMEA one (and in the u-blox's case the NMEA config was invalid and would not work anyways).

Probably best to just bring the MTK/SIRF blobs back. Testing locally shows that adding the MTK driver back in is 2.5k, which is far more memory then we save by deleting the MTK blobs :)

night-ghost, 2017-10-17:
it will be better to not delete commonly used features without explicit testing and if it REALLY required deprecate that feature for some time.

WickedShell, 2017-10-23:
As was stated it was not an intentional breakage of the feature. It wasn't being tracked that anyone would be using MTK gps's in NMEA mode.

The current proposal is to remove the MTK driver entirely, and special case the `GPS_TYPE` 3 or 4 (MTK/MTK19) to actually just use the NMEA driver and send the removed config blob on startup. The current NMEA driver provides all the features that the MTK drivers did, so this should provide more options for future MTK users, while not costing old ones significant features.

I don't have any MTK units here to test with, but would you have any concerns (and be able to test the changes if someone else implements it) about this approach @night-ghost  ?

night-ghost, 2017-10-24:
I think this will be the good solution because NMEA is a worldwide standard where binary protocols can be changed in new chipsets. But there will be only a slight sadness that all was working in the "auto" before.
I use ONLY GPSes on MTK chipset (UBLOX can cause flyaways) so of cause will test it. For my needs I just reverted that commit, though

iddq, 2020-09-17:
where could I find mtk binary protocol description?

WickedShell, 2021-09-29:
Officially removed all support in #18785 so I'm going to go ahead and close this.

closed 2021-09-29 by WickedShell