ArduPilot/ardupilot #7082: Pixhawk with APM v3.5.3, Frsky S.Port telemetry

zs6buj, 2017-10-17:
#### Issue details
Hi Guys

Forgive me if I am wrong, but with the SERIAL4_PROTOCOL parameter set = 4  (in Mission Planner), I should be getting new S.PORT protocol telemetry, but I'm getting the old D hub style telemetry. I searched for reports on similar issues without success.
 
With the parameter set = 10 I get Frsky Passthrough - no problem.
With the parameter set =  4 I get the D style telemetry shown in the attachement
With the parameter set = 3 I also get D style telemetry, but seems to be less verbose.

I'm expecting two byte value types like in the SPort definition, but get 1 byte values types like 0x1, 0x5, 0x10, 0x21 and so on.

I'm using the telemtery, which gets inverted and goes to an XSR receiver, then down to ground Taranis and is presented on the tx pin in the back bay of the Taranis, for a tracker. It's working fine with D style telemetry, if the legend says 4 = S.Port then it should be correct, don't you think?

[image]

[1 Pixhawk serial4 output.txt](https://github.com/ArduPilot/ardupilot/files/1390566/1.Pixhawk.serial4.output.txt)
[2 Frsky Telemetry D and X codes and structure.txt](https://github.com/ArduPilot/ardupilot/files/1390567/2.Frsky.Telemetry.D.and.X.codes.and.structure.txt)

#### Version
Pixhawk APM v3.5.3

#### Platform
[  ] All

[  ] AntennaTracker
[x  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Hex F550

#### Hardware type
Pixhawk

#### Logs
No logs, but see attachements

OXINARF, 2017-10-20:
@floaledm Can you please answer this?

@zs6buj I'll let it slide this time, but this issues list is for feature requests and confirmed bugs. For user support please post in the forum (http://discuss.ardupilot.org).

floaledm, 2017-10-20:
@OXINARF  Thanks Francisco.

@zs6buj Thanks for your post. According to the Smart Port protocol spec (I have v3.3.2), what is referred to as the old DATA IDs (a.k.a. Application IDs, sensor IDs) which are 1 byte long are referred to as "version 1/old sensors in the FrSky Smart Port spec, but nothing indicates that it's specifically "D style telemetry." Do you have links to the SPort definition that shows SPort telemetry is necessarily 2 bytes long for IDs?

Is there anything broken/not working (I'd like to understand the reasons behind the requested change)? OpenTX recognizes both the 1-byte and 2-byte IDs, so that part should be fine. What is the tracker you're using that apparently recognizes only 2-byte IDs? I support the change to 2-byte IDs if you'd like to issue a PR to that effect, and I'd be happy to review it.  The other option is for the tracker to be programmed to recognize 1-byte IDs.

Let me know what you find out.

zs6buj, 2017-10-21:
Thank you for your reply Francisco,
The tracker is not important as I simply coded for both "old" 1 byte and new 2 byte id's.
The issue is that Frsky no longer make or sell D style receivers. As far as I am aware all new receivers are branded X and equipped with the S.Port, and their native protocol uses the new 2 byte I'D.
Besides that, the Ardupilot parameter mentioned above clearly caters for the old D hub system when = 3.
If parameter = 4 also supports only the old 1 byte ID's, AP has no support for 2 byte X protocol, which represents the future.
Yes there is also "passthrough", which uses 2 byte Id's in the 5000 range (except for gps data), but many sensor types are omitted for that spec presently.

Thanks again for responding.
Eric

zs6buj, 2017-10-21:
Francisco, take a look at the attached frsky.h header file used by OpenTX. The "new" 4 byte ID record types clearly duplicate the old 1 byte types, and frankly are far more sensibly coded.  I also understand that the protocols (IDs) can and do co-exist, but the new ones are smarter. I also gather that Frsky, the vendor/manufacturer, has been reticent in recent times of publishing their proprietary protocol, perhaps understandably.

All the above notwithstanding, I must admit I am new on the Frsky scene (I bought a Taranis :)), and after a lot of research this is my take on the state of play. There will be many people out there that know a lot more about this, including the history as the protocol evolved. 
[frsky.zip](https://github.com/ArduPilot/ardupilot/files/1403884/frsky.zip)

floaledm, 2017-10-21:
@zs6buj I'm not trying to nitpick but simply attempt to straighten the record here, so please bear with me...
> As far as I am aware all new receivers are branded X and equipped with the S.Port, and their native protocol uses the new 2 byte I'D.
Yes, the IDs can be up to two bytes long, but using the 1-byte IDs is not incorrect as far as I understand and complies with SPort protocol spec (it's in that document). Also, the 1-byte IDs work just fine on the X series (i.e., you don't need D series to use protocol 4). You may want to revise your coding of the "old" version to support both the 1-byte and 2-byte IDs (on the X series, a 1-byte ID is a 2-byte with one of the bytes always at 0x00). This way, you should be able to get protocol 4 to work with both D and X series.

> Yes there is also "passthrough", which uses 2 byte Id's in the 5000 range (except for gps data), but many sensor types are omitted for that spec presently.
Passthrough is part of the SPort spec (even though not published). This is something that was coordinated between the OpenTX devs and FrSky. Addresses in the 0x5000 range are reserved for data that is specifically not sensors (at least in the sense that they should be directly "discoverable" by OpenTX)! I understand there is not a lot of info about this out there, but you can ask questions. We have to be careful not to misrepresent the facts...

> If parameter = 4 also supports only the old 1 byte ID's, AP has no support for 2 byte X protocol, which represents the future. [...] The "new" 4 byte ID record types clearly duplicate the old 1 byte types, and frankly are far more sensibly coded.
We have to be careful though not to try to fix what isn't broken, which in turn could break the functionality for some users that right now use protocol 4 the way it is. As I said I support the change to 2-byte IDs, and actually for similar reasons as you ("represents the future" and "the new ones are smarter"), but there's work that should be done to to figure out if that change to 2-byte IDs would break anything of significance since there's actually no compelling reason to move to 2-byte IDs...

Bottom line, you may simply want to update your tracker code to be compatible with v1 and v2 of the SPort spec. Both the 1-byte and 2-byte IDs work fine on the X series (as long as the 1-byte ID is coded on 2 bytes, as is done in ArduPilot).

OXINARF, 2017-10-22:
> on the X series, a 1-byte ID is a 2-byte with one of the bytes always at 0x00
> as long as the 1-byte ID is coded on 2 bytes, as is done in ArduPilot

I think these are the most important statements: the protocol 4 in ArduPilot uses 2-byte. And, as allowed by the protocol, it uses the old IDs instead of the new ones.

@zs6buj It's clear to me now that this isn't an issue so I'm closing it. If you have more questions to get your tracker working, please post in our forum (http://discuss.ardupilot.org).

closed 2017-10-22 by OXINARF

zs6buj, 2017-10-22:
Thanks guys for filling in the gaps for me. I understand that changing code may well affect others, and agree with the conclusion to close the call.