ArduPilot/ardupilot #7088: Possible bug in sending heartbeat message

SalahuddinAsh, 2017-10-17:
#### Issue details
When I arm the vehicle from my ODroid companion computer, the received heartbeat message states that the vehicle is armed. When the vehicle is disarmed, one received heartbeat message state that the vehicle is armed, while the next message states that the vehicle is disarmed, and so on. 

Also, when I set the mode to be any thing but STABILIZE, say GUIDED, one received heartbeat message states that the custom_mode is the desired mode, e.g., GUIDED, while the next message states that the custom_mode is STABILIZE.

I'm connecting my pixhawk to QGroundControl. QGroundControl gives the right status of the vehicle, i.e., whether the vehicle is set or not, and what is the current flight mode.

I tried to investigate that problem with no success. 

Note: I tested my code, which is on ODroid, in simulation and it worked fine without having this problem.

#### Version
Copter: 3.4.6, 3.5.1, 3.5.3

#### Platform
Copter

#### Airframe type
Y6

#### Hardware type
pixhawk

#### Logs
_Please provide a link to any relevant logs that show the issue_

peterbarker, 2017-10-17:
On Tue, 17 Oct 2017, SalahuddinAsh wrote:

> I'm connecting my pixhawk to QGroundControl. QGroundControl gives the right
> status of the vehicle, i.e., whether the vehicle is set or not, and what is
> the current flight mode.

Check the vehicle dropdown in QGroundControl.  Is it showing multiple 
vehicles?

SalahuddinAsh, 2017-10-17:
@peterbarker Thanks for your reply.
With "vehicle dropdown" you mean the map, right? It's displaying "No GPS Lock for Vehicle"

peterbarker, 2017-10-18:
On Tue, 17 Oct 2017, SalahuddinAsh wrote:

> @peterbarker Thanks for your reply.
> With "vehicle dropdown" you mean the map, right? It's displaying "No GPS
> Lock for Vehicle"

Mmm... no.  QGroundControl has pretty nice multiple-vehicle support.  I 
have a theory that you're seeing multiple heartbeats from different 
system ids.  A lot of things send heartbeats - includings GCSs - and some 
things get confused when they don't filter to a single system id when 
displaying information derived from the telemetry stream.

I thought looking in the QGrundControl interface to see if it saw two 
vehicles might be a quick check.

SalahuddinAsh, 2017-10-18:
So, is it possible that my companion computer receives heartbeat messages from more than one system, although it is physically connected to only Pixhawk?
As I said in above, QGroundControl displays consistent status, when the vehicle is armed, it displays armed, and vice versa. But my program on the companion computer displays conflicting status, if the vehicle is armed, it displays armed, but when the vehicle is disarmed, it receives one heartbeat message as armed, and the next as disarmed. 
Here is a post I made beforehand with my code and more details https://discuss.ardupilot.org/t/copter-conflicting-received-heartbeat-messages/22102

peterbarker, 2017-10-18:
On Wed, 18 Oct 2017, SalahuddinAsh wrote:

> As I said in above, QGroundControl displays consistent status, when the
> vehicle is armed, it displays armed, and vice versa. But my program on the
> companion computer displays conflicting status, if the vehicle is armed, it
> displays armed, but when the vehicle is disarmed, it receives one heartbeat
> message as armed, and the next as disarmed.
> Here is a post I made beforehand with my code and more detailshttps://discuss.ardupilot.org/t/copter-conflicting-received-heartbeat-messa
> ges/22102

My guess is that your program is seeing heartbeats from QGroundControl. 
Print out the system id ofthe heartbeat packet and check it against tha 
tconfigured in QGC.

SalahuddinAsh, 2017-10-18:
Man, you are genius. It was really the problem as you said. 
I found that my program receives messages from two systems. One with ID of 1 (which is Pixhawk), and the other with ID of 255 (which I don't know what is it).
After considering only the messages come from ID 1, the problem vanished. 
Could you please explain, what is the system with ID 255, and how can my companion computer receive messages from another system while it is not physically connected to it?
Million thanks

peterbarker, 2017-10-18:
On Wed, 18 Oct 2017, SalahuddinAsh wrote:

> Man, you are genius. It was really the problem as you said.
> I found that my program receives messages from two systems. One with ID of 1
> (which is Pixhawk), and the other with ID of 255 (which I [...]

Without seeing a diagram of how you have things connected, I'm 
speculating.  But it's probable the PixHawk is relaying the telemetry from 
one of its mavlink connections to another.

As mentioned in the previous post - check QGC's system id, which is in its 
configuration under connections, I believe.

Note that it is important that a group of systems which can see each other 
should all have different IDs.  Things get a little confusing otherwise 
:-)

> Million thanks

No worries :-)

OXINARF, 2017-10-20:
@SalahuddinAsh This issues list is for feature requests and confirmed bugs. For development questions please post in the forum (http://discuss.ardupilot.org) or ask in Gitter chat (http://gitter.im/ArduPilot/ardupilot).

ArduPilot routes most MAVLink packets, unless they are addressed specifically to the autopilot - and many MAVLink packets don't have a destination, are just supposed to be broadcast, which is what ArduPilot does with them.

closed 2017-10-20 by OXINARF