ArduPilot/ardupilot #7090: Circle_Radius change in flight via Pitch Stick

RJBeckwith, 2017-10-18:
Request option that Circle_Radius be adjustable in flight via the Pitch stick when in circle mode for current version of Arducopter.  Circle rate should remain adjustable via CH_6 option.

The 3DR Solo uses the Pitch control for adjusting radius in flight, which works very well.

Pedals2Paddles, 2017-10-18:
That is actually a great idea. The pitch input has no other function in circle mode and this would fill a significant gap in that mode's practicality.  And on that note, it might even be worthwhile to adjust the circle_rate with the roll stick.  Then you don't have to give up CH6 gimbal pitch control.

RJBeckwith, 2017-10-18:
I really like the roll idea! Perhaps even allowing you to change direction as well as speed?  As you pointed out, the sticks have no other use in circle mode and input is normally ignored.  I really don't want to give up my CH_6 options!

OXINARF, 2017-10-21:
Duplicate of #5450

closed 2017-10-21 by OXINARF