ArduPilot/ardupilot #7091: Ardupilot: Unable to detect RPI version

lvale, 2017-10-18:
#### Issue details
When starting ArduPilot (any version and vehicle) under the new Debian Stretch the correct type of RPI is not detected
#### Version
all
#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine
#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
Linux+RPI2/RPI3
#### Logs
_Please provide a link to any relevant logs that show the issue_

```
pi@navio2:~ $ arducopter --help
Could not detect RPi version, defaulting to 2/3
Usage: -A uartAPath -B uartBPath -C uartCPath -D uartDPath -E uartEPath -F uartFPath
Options:
```

lucasdemarchi, 2017-10-21:
What's the output of /proc/cpuinfo?

lvale, 2017-10-22:
```
pi@navio2:~ $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 57.60
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 1
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 57.60
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 2
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 57.60
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 3
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 57.60
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

Hardware	: BCM2835
Revision	: a01041
Serial		: 00000000fcd20a99

```

lucasdemarchi, 2017-10-22:
We are missing `Hardware	: BCM2835` in our RPI version check.

lvale, 2017-10-22:
pi3
Hardware	: BCM2709

lvale, 2017-10-22:
pi0
Hardware        : BCM2708

lvale, 2017-12-05:
on a rpi3

```
pi@navio:~ $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02082
Serial		: 000000000153c74b

```

lucasdemarchi, 2017-12-06:
In our codebase we are handling BCM2709 (RPI 2) and BCM2708 (RPI 1). Everything else is handled as BCM2709.

I think the only place we really care now about those differences is in RC_Input since we are accessing HW registers directly. In other cases we are using the HW abstraction. And this is only true for some extension boards, not all.

What board are you using with RPI?  Is it working and just printing an annoying message or is it not working at all?

lvale, 2017-12-07:
on the Emlid Navio2 board there are no issues with RC.

I believe #7369 corrects this

closed 2017-12-07 by lvale