ArduPilot/ardupilot #7093: Parrot Bebop "make bebop" Command Fails

ericjackson613, 2017-10-19:
Hello,

I saw this mentioned as a failure on a previous post from 2 years ago (https://github.com/ArduPilot/ardupilot/issues/3208) and I followed that solution but am still running into the same problem.

I am trying to get the Bebop up and running with ArduCopter, but one step is giving me issues. I have been following along with the ArduPilot instructions for building it on Linux here: http://ardupilot.org/dev/docs/building-for-bebop-on-linux.html

I have successfully updated the firmware on the Bebop to be compatible with ArduPilot and am now attempting to build ArduCopter for the Bebop. I am working with the original Bebop. I installed the armhf toolchain, installed git, and cloned the ardupilot repository. I have been following the steps to build the flight control hardware as if I would build it for the Pixhawk (http://ardupilot.org/dev/docs/building-px4-for-linux-with-make.html#building-px4-for-linux-with-make1). After changing the directory to the ArduCopter folder in the git repository, I attempt to build it specifically for the Bebop by using the command **make bebop** but get an error that says: “make: *** No rule to make target ‘bebop’. Stop.” When I look in the Makefile in the git repository, I cannot find a target which will allow me to create the flight control firmware specifically for the Bebop.

Has anyone else encountered an issue like this or has a possible solution? It might be that the site I've been following is outdated. I’m still learning Unix and am new to how the make command and Makefile work so if seems like an issue with either of those, I would love any insight that you have.

This is what my terminal reads when the problem arises: 

**sentien@c3dmac-sid:~/ardupilot/ArduCopter$ make bebop
make: *** No rule to make target 'bebop'.  Stop.**

Thank you,
Eric

jberaud, 2017-10-19:
Hi @ericjackson613 
The wiki is a bit outdated.
Now the instructions for Bebop 2 should work on bebop 1.
Moreover, in order to build, you need to use waf:
./waf configure --board=bebop --static
./waf build
Normally you don't have to build statically on Bebop2 anymore. On bebop 1, I have to check a bit further.

Regards,
Julien

OXINARF, 2017-10-21:
This issues list is for feature requests and confirmed bugs. For development questions please post in the forum (http://discuss.ardupilot.org) or ask in Gitter chat (http://gitter.im/ArduPilot/ardupilot).

Make has been removed for Linux boards. Unfortunately developer documentation hasn't been updated yet, but you can read all about waf (our new build system) at https://github.com/ArduPilot/ardupilot/blob/master/BUILD.md

closed 2017-10-21 by OXINARF