ArduPilot/ardupilot #7095: In Auto/Missions: Motor failure to be detected, to disable Yaw and to activate RTL

KirkKirk, 2017-10-20:
#### Issue details
Recently during a flight in Auto (Mission) I lost my Hexa because one motor failed.
In the last 15 days, after a lot of analysing, I think that every Hexa will survive if a few
things **can be identified/detected**
- when sudden high vibrations pop up
- when there is 100% throttle on the one motor but very low throttle (or "0%") on the opposite motor
- when there is detect high HUD activity (high wobbling, flips etc.)
**and after that the RTL to be activated and the Yaw ("the killer") to be disabled**

I am willing to help with testing it in the safe environment (after I finish building my new hexa with Pixhawk 2.1 in a few weeks) 

#### Version
AC 3.5.2

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Hex

#### Hardware type
Pixhawk 1

#### Logs
No logs present as the drone has been lost in the ocean. (I have only the MP .tlog file)

magicrub, 2017-10-20:
Please post logs, this should already be implemented. Success depends on if the battery and other motors are able to handle the added load. Please see the following videos for reference:
https://www.youtube.com/watch?v=FNQgdcImpKs

https://www.youtube.com/watch?v=ap8ngQqt7s8

OXINARF, 2017-10-21:
Duplicate of #838.

closed 2017-10-21 by OXINARF