ArduPilot/ardupilot #7096: PX4Flow with Intel Aero RTF

SubhiH, 2017-10-20:
#### Issue details
Connecting PX4Flow sensor with Intel Aero RTF that uses Ardupilot flight stack.
When I connected the sensor with Telemetry port as this [article](http://www.instructables.com/id/Intel-Aero-Drone-Altitude-and-Position-Hold-Using-/) shows, I was not able to get the data from the sensor and whenever I enabled the sensor, It was disabled automatically. When I connect the sensor with Compass board, It works without any problem.

#### Platform
Copter

#### Hardware type
Intel Aero RTF

magicrub, 2017-10-20:
@SubhiH sorry this is taking so long to get a solution, I'll follow up!

SubhiH, 2017-10-21:
@lucasdemarchi has figured out the solution, the drone should have the latest fpga firmware so that the PX4Flow sensor can work with telemetry port in Intel RTF.

closed 2017-10-21 by SubhiH

magicrub, 2017-10-21:
Also see https://m.facebook.com/groups/1068587753164417?view=permalink&id=1595780907111763 for additional discussion

OXINARF, 2017-10-22:
I'm failing to see how this is an ArduPilot problem:

- the linked guide isn't for ArduPilot
- the linked guide has a custom FPGA configuration
- we follow the FPGA configuration provided by Intel, not custom ones, so quite likely the telemetry port is configured as a serial port
- for custom configurations, a custom firmware is going to be needed (as the linked guide also uses)
- @SubhiH says it works on the I2C port as it should