ArduPilot/ardupilot #7099: AP_Beacon_Marvelmind: Low memory when allocating can lead to dereferencing nullptr

WickedShell, 2017-10-20:
#### Issue details
When allocating the `AP_Beacon_Marvelmind` instance, an instance of `MarvelmindHedge` needs to be allocated as [well](https://github.com/ArduPilot/ardupilot/blob/2fcb4932b70f9a851731d3ab9c4d543b6e944810/libraries/AP_Beacon/AP_Beacon_Marvelmind.cpp#L33), this can fail however, but AP_Beacon is unaware that the allocation of the hedge object failed. The rest of `AP_Beacon_Marvelmind` is unaware of the possibility that the hedge object was not allocated, and will happily dereference a nullptr.

AP_Beacon needs to check that the `AP_Beacon_Marvelmind` instance actually managed to create it's members and delete it if not. (The easiest way is to probably move to a `bool AP_Beacon_Marvelmind::init(void)` approach which if it fails AP_Beacon can use to indicate that the driver should be deleted.

#### Version
Master

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Hardware type
Marvelmind

magicrub, 2017-10-20:
Was this found by Coverity?

WickedShell, 2017-10-20:
Kinda, it was complaining that there wasn't a destructor (see CID 144334) for it, I just noticed the rest looking at it. The reality though is that it turns out `MarvelmindHedge` is just a struct and will never be recreated. It should just be a member of the `AP_Beacon_Marvelmind` object and this whole issue goes away, and all the pointer dereferencing is cleaned up... And a ton of the fields in that struct are actually compile time constnats that should live in a #define rather then in a struct.

amilcarlucas, 2017-10-23:
Thank you Michael, I will try to look at this in the next few days.

ptsneves, 2017-10-31:
It is ironic that I had a good patch for this, but it was ignored until a bug was raised. The code was poor with nullptr problems being the most obvious consequence.

closed 2017-11-01 by rmackay9

amilcarlucas, 2017-11-01:
It was ignored because you ignored the review coments from Francisco and me.

magicrub, 2017-11-03:
related to: https://github.com/ArduPilot/ardupilot/pull/6474

PR #7154 (2017-11-01): AP_Beacon: Corrected possible use of nullptr memory.
This is a subset of #6474 with a couple of small changes from me. @ptsneves I hope you are OK with this
It should close #7099