ArduPilot/ardupilot #7103: Planned mission disappears when entering a guided go-to location

fredrikfalkman, 2017-10-21:
When flying a mission in auto, if I enter a go-to location (Guided), the mission disappears from the chart screen. If I go back to Auto, the plane reverts to following the planned mission, but the waypoints and route lines are still not showing.

Other than that, the new ability to use go-to locations seems to work nicely. Much appreciated!

3.8 (Development add505fa7 2+17-09-17)

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Flying wing

#### Hardware type
RPi3/Navio2; Arduplane 3.8.1

OXINARF, 2017-10-22:
@fredrikfalkman How is this related to ArduPilot? Looks like a GCS problem.

magicrub, 2017-10-22:
Exactly. This is a UI issue and I assume he's talking about mission planner. If so, please resubmit this to the mission planner issues page here
https://github.com/ArduPilot/missionplanner/issues

closed 2017-10-22 by magicrub

fredrikfalkman, 2017-10-22:
Yea, sorry, wrong forum, too many browser windows open...  resubmitting in the QGroundControl forum.