ArduPilot/ardupilot #7113: Throttle issues in guided mode

JamesMegariotis, 2017-10-23:
#### Issue details
When using the `SET_POSITION_TARGET_GLOBAL_INT` message in guided mode, the throttle exhibits some strange behavior. I have noticed a few issues related to performance when using this message for setting the position and velocity of the UAV. In Copter firmware (Copter 3.5.3, and all other recent versions including Master), there appear to be very violent throttle spikes and desired altitude discontinuities when setting position and velocity with a change in altitude. It behaves as expected when sending only position with an altitude change. It behaves as expected when sending only x and y position changes with x and y velocities, but when velocity is sent with a new position that includes a change in z position, these issues occur. The aircraft does not obey acceleration parameters and maxes out the throttle for 1 time step. On descent, the throttle goes to zero. I have been searching the code in the guided flight mode and the attitude control and wp_nav libraries. I have only been able to determine that this has to be related to how the pos control's z controller is updated `pos_control->update_z_controller()` when it encounters a change in altitude in the guided flight mode's posvel_control mode. It is not present when using only pos_control (aka wp_control). One other interesting thing to note is that the discontinuities in desired altitude are equal in magnitude and direction to the values of `wp_nav->get_speed_down()` and `wp_nav->get_speed_up()`. It appears to me that the cause of this problem is due to the fact that the desired climb rate is not smoothed when set, as can be seen in the plot below. The plots below come from a replication of this issue using sitl simulation. You can see that when an alt change is requested, the desired climb rate steps up to wp_nav speed up or down and the desired altitude steps up by the same amount. This causes the violent throttle spike and the vehicle does not obey any limits placed on it. Interestingly, this problem can be circumvented by ensuring your vehicle is already traveling at the desired climb rate using velocity control before switching to posvel control. This eliminates the throttle spikes by ensuring the vehicle obeys the acceleration limits, but is not a long term solution. Any help would be greatly appreciated. I would be more than happy to provide the logs or anything else that would be helpful. I would also be more than happy to share my thoughts on the cause of the issue and areas that I have explored and attempted to resolve this issue. I have also noted some smaller issues associated with velocity control and how altitude error is subtracted from the desired climb rate, making it impossible to reach the velocity requested, but may be worthy of opening another issue. 

**Throttle Plot**
[image]

**Altitude Plot**
[image]

**Climb Rate**
[image]
Note the difference between the way takeoff handles the climbrate, and how the posvel messages cause a step in desired climb rate.

**Notes on Above Plots**: The x axis showing line numbers is not scaled the same in all three photos. The alt plot was scaled and cropped so as to make the discontinuities clearly visible.

#### Version
All versions. Sitl log comes from Copter 3.5, but can be demonstrated on any version utilizing the pos control library.

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Any multicopter frame. Issue has been observed in hex, octa, octa-quad, and quad. Also in sitl quad.

#### Hardware type
Pixhawk 2.1 and Pixhawk

#### Logs
I would be happy to provide the log files once I am in the office. I am submitting this issue from home but I have extensive examples of log files where the bug is present. I have provided log plot images of the relevant flight variables.

OXINARF, 2017-10-24:
@JamesMegariotis Please do provide the logs. Also, please try to format the initial paragraph a little bit better, it's a huge block of text, very hard to read.

Usually issues are addressed in Gitter or the forum, but @JamesMegariotis has already talked about it in Gitter with no resolution so, because it looks like a real code issue to me, I'm allowing it here.

JamesMegariotis, 2017-10-26:
I have attached a couple logs that show the throttle spikes. These are from real flights on a hexacopter. I have more including sitl logs as well. Let me know how I can help.

And sorry for the poorly formatted message. this is my first time posting an issue.

[2017-10-10 11-39-41v353posvelspikes.log](https://github.com/ArduPilot/ardupilot/files/1419662/2017-10-10.11-39-41v353posvelspikes.log)
[2017-10-10 16-19-18v353posvelspikes2.log](https://github.com/ArduPilot/ardupilot/files/1419663/2017-10-10.16-19-18v353posvelspikes2.log)

OXINARF, 2019-08-20:
@JamesMegariotis is this still relevant on master? We had some corrections on this area, not sure if they fixed this or not.

rmackay9, 2019-08-21:
My understanding is that the velocity was being provided which would make it use the pos-vel controller instead of the regular pos controller.  It's not a commonly used controller and I suspect the sender was incorrectly sending the final desired altitude.  Instead the sender needs to ramp up/down the requested positions and velocities.  The sender should be slowly increasing the desired velocities and positions so that the two are consistent with each other.

JamesMegariotis, 2019-08-26:
@rmackay9 This issue occurs when using the guided pos controller or guided pos-vel controller via SET_POSITION_TARGET_GLOBAL_INT. My workaround currently uses the guided velocity controller in order to reach the velocity setpoint before switching to pos-vel control or pos control. The velocity controller seems to obey acceleration limits or some sort of slew-rate limiting on throttle and does not exhibit this "throttle spike" behavior you see when using either position control or pos-vel control. Your suggestion of manually ramping the velocity in pos-vel could work but I doubt it. The "throttle spikes" on altitude changes happen regardless of the magnitude of commanded velocity in pos-vel control mode and happen on any SET_POSITION_TARGET_GLOBAL_INT message where the altitude change is greater than some seemingly arbitrary threshold that may be related to leash length (or some other alt change threshold that I am unaware of). I am away at the moment but I will test this on master as soon as I get back next week.

@OXINARF I will attempt a repeat of previous tests on the current master once I am back in the office to verify that this issue is still present. It is worth noting that this behavior occurs in SITL, so verifying this behavior can be done using simple SET_POSITION_TARGET_GLOBAL_INT message. I have verified this using the 3.5 and 3.6 branches of arducopter. I have not tried on master or at least not in the past year.

OXINARF, 2019-08-26:
@JamesMegariotis If this is still reproducible, please provide the steps for SITL, makes it easier to look at it. Thanks!

OXINARF, 2019-08-26:
@JamesMegariotis No hurry! I just sometimes see old issues and try to understand if they are still valid (we unfortunately have a huge list).

IamPete1, 2021-08-15:
This has almost certainly been fixed in the recent SCurves rework. Please reopen if the issue is still present on 4.1.

closed 2021-08-15 by IamPete1