ArduPilot/ardupilot #7119: EKF altitude measurement failure

RoelHelsen86, 2017-10-24:
#### Issue details
when I enable the EK2_RNG_USE_HGT parameter to 10, with a rangefinder max altitude of 15m and I fly away in loiter for some 100 meters or so, then switch to RTL, the copter comes back, waits the set loiter time and then starts descending. At this moment, the EKF switches and the altitude is completely wrong (stays more or less stationary). This results in the copter descending at considerable speed as it tries to attain the desired descend speed.

For some reason the EKF does not want to use the barometer (barometer innovations are very high anyway) and the lidar is still above its maximum altitude, so I assume that its values are not used. This problem only occurs when EK2_RNG_USE_HGT is on. I assumed that in this case, the EKF would only use the rangefinder altitude below the set percentage of the rangefinder max altitude.

can someone tell me what the EK2_RNG_USE_HGT enables and why it makes the EKF crash?
I'll take a look at the code myself as well, but I am not (yet) familiar with all the EKF code.

The problem is that the altitude measurement of the EKF becomes completely erratic and thus any mode that uses altitude hold is unusable. It only occurs when EK2_RNG_USE_HGT is enabled and only if I fly away far enough from the launch point, but in these conditions it happens consistently.

I attached a log file from a flight where this happens (last flight of the log)
[image]

#### Version
Copter 3.5.3

#### Platform
[  ] All
[  ] AntennaTracker
[ x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
X quad

#### Hardware type
pixhawk 2, using leddar_one rangefinder

OXINARF, 2017-10-24:
I'm going to leave this open for a bit until you attach a log, otherwise I'll close it because this issues list is for feature requests and confirmed bugs. For user support you need to post in the forum (http://discuss.ardupilot.org).

RoelHelsen86, 2017-10-25:
[image]
Thanks for the answer,

I tried uploading the log by pressing the "selecting them" text and then selecting the appropriate file and did this again in this post. I selected the .BIN file that comes from the SD-card in the pixhawk. If this does not work, could you please explain me how to add the correct log.

I reported it as an issue, because in my opinion the ekf crashes leading to a dangerous situation. I wanted to report this as a bug, because I think that the firmware is doing something weird/wrong. It might also be the case that I'm asking/expecting the wrong things from the ekf, in which case the topic should indeed be moved to the forum.

RoelHelsen86, 2017-10-25:
[log.zip](https://github.com/ArduPilot/ardupilot/files/1413645/log.zip)

This might be the right way of doing it, the added file contains the log I wanted to upload with my previous posts. The log contains several flights where we were testing the rtl function. for the last flights we were flying a bit further away each time to trigger the issue. This happened in the last flight of the log. If this is not enough data, I can make new flights to gather more logs.

RoelHelsen86, 2017-10-25:
[image]

Here is also a picture of the log indicating what/where the issue is

priseborough, 2017-10-28:
The log you have attached has EK2_RNG_USE_HGT set to -1. 

Also note that EK2_RNG_USE_HGT specifies the maximum percentage of the range finder max range that will be used and that when coming back within range it doesn't start using it until 70% 

If you have been using  EK2_RNG_USE_HGT = 10 that means then it won't start using range finder on descent until 1 metre, which is much too late. 

Please use EK2_RNG_USE_HGT = 70 and retest.

OXINARF, 2017-11-05:
@priseborough The log only has EK2_RNG_USE_HGT set to -1 right in the end, the issue happens before that when EK2_RNG_USE_HGT is set to 10.

tizianofiorenzani, 2018-03-09:
I just wrote a post on the forum about it. I tested today and almost crashed. When the altitude is hel, it does nicely, but the transitions are awful.
 Check my post here: https://discuss.ardupilot.org/t/copter-3-5-2-ekf2-bad-behavior-with-rangefinder-enabled/21595/3

[image]

As you see, as soon as the rangefinder was activated (setting the EK2_RNG... = 100) the altitude dropped significantly from 4 meters to almost 1 (the sonar altitude is how close was to the ground). The EKF altitude was completely wrong, and jumped up as soon as I activated the feature.

[image]

I have the bin if anybody wants to have a look to the data

priseborough, 2018-03-09:
Please attach a log and note that the feature should not be activated or deactivated in mid-flight as this use case has not been tested.

tizianofiorenzani, 2018-03-09:
I uploaded the file on my drive: https://drive.google.com/file/d/1U-J2UVwbyYD0jknDBjutva2_1KmjU-h7/view?usp=sharing

festlv, 2018-06-09:
I am not sure if this problem is the same but I have a reproducible test case for a very similar issue- EKF PD estimate becomes stuck when EKF fuses rangefinder.

I have a spectacular log file of this problem here: https://linx.wot.lv/dmbax127.bin

Illustration of the issue:
[image]

At height of 14m vehicle received an RTL command (to 15m height). Vehicle started climbing, but EKF PD estimate became "stuck". Vehicle then climbed to a height of 2.5km (according to barometer, 2.3km according to GPS). At that point EKF it seems that EKF switched to second IMU and PD estimate became correct. However, after descending a few hundred meters the crash detector kicked in and vehicle disarmed (at a height of more than 2km). I'm very glad that we were able to find the vehicle and retrieve the log.

Initially, this issue was very puzzling: all the sensors were reporting correctly, no crazy vibrations.
I tried replaying the log in Replay tool and according to Murphy's Law, EKF PD estimate was correct there.

However, Replay does not feed rangefinder data to EKF. Once I added rangefinder to Replay, I was able to reproduce the EKF PD estimate getting stuck. Here's my quick and dirty patch to add rangefinder, based on release 3.5.3: https://gist.github.com/festlv/52168ad69eab5ac1c043ed92149c2b5d

On a closer investigation, it appears that EKF fused rangefinder data and completely ignored barometer and GPS, although the data from rangefinder was outside the range of RNGFND_MAX_CM and consequently, EK2_RNG_USE_HGT.

Although the issue can be considered user error (when setting RNGFND_MAX_CM to 2000, which is our rangefinder's range the PD estimate is correct), I don't think it's the expected behaviour from EKF.

rmackay9, 2018-06-11:
I'm surprised that people are modifying the EKF parameters related to the range finder (EKF_ALT_SOURCE or EKF_RNG_USE_HGT).  I don't think this is ever a good thing to do and it's not mentioned anywhere on the wiki as far as I know.  I really think those parameters should always be left at their defaults.

Perhaps we should add some comments to the wiki and the parameter description to discourage people from modifying them.  Maybe we should remove them all together.

@priseborough what do you think about my suggestions above?  am I misunderstanding when people should modify these parameters?

WickedShell, 2018-06-11:
@rmackay9 I can't speak to the RNG_USE_HGT but there are a lot of good reasons to mess with ALT_SOURCE and we shouldn't be actively steering people away from it. (Better explanations sure, but we shouldn't be making it taboo, particularly worth noting setting the GPS as alt source can be a very good idea).

rmackay9, 2018-06-11:
@WickedShell, I definitely agree that setting EKF_ALT_SOURCE = GPS has a very valid use case. 

.. but the use for setting EKF_ALT_SOURCE = RangeFinder is only when flying indoors in an environment where the floor is nearly perfectly flat with no obstacles below (a very rare case).

.. I think EKF_RNG_USE_HGT should only be set to a positive value when the EKF_ALT_SOURCE = RangeFinder.

priseborough, 2018-06-11:
Use of EK2_ALT_SOURCE to switch to range finder when low and slow was developed for a special use case where the Baro could not be used close to ground due to high prop-wash and was only intended to assist during vertical takeoff and landing with the copter climbing above the maximum use height before flying horizontally. Unfortunately the limitations were not made clear in the parameter documentation (my fault) with the result that it is being used for general flying and terrain following.

The altitude limit must be set to be within the range finders published maximum, I would never set it more than 70 to allow for sensor degradation and with a sensor that has only 2m of range it is doubtful it can be used reliably at all because there will not be enough time for the EKF to switch if the baro and terrain height datums have drifted relative to each other.

If someone wants to use range finder all the time, then EKF_ALT_SOURCE = RangeFinder should be used, but not in combination with EKF2_RNG_US_HGT.

festlv, 2018-06-11:
The rationale for using EK2_RNG_USE_HGT was to improve height estimate when hovering in windy conditions, close to ground. 

In the particular case I described above, it appears that EKF switched the alt source to rangefinder, even though it's measurement was outside RNGFND_MAX_CM (700) * EK2_RNG_USE_HGT (50) = 3.5m.
Rangefinder measured 20m at that point- see the plots below.
[image]

tizianofiorenzani, 2018-06-11:
In my case, I provide a rangefinder through mavlink protocol. The reason why someone would switch on and off the rangefinder is because it does not look trustworthy and we need to deactivate in case it goes nuts. I have tried and leave it active in the EK2 and giving zero as signal, but it would interpret the signal as valid and think is on the ground... There is definitely something odd. Have not tried on the latest 3.6 though

priseborough, 2018-06-11:
@festlv I'm reviewing the https://linx.wot.lv/dmbax127.bin log and code now to try and understand why it started using the range finder.

priseborough, 2018-06-11:
It looks like the terrain estimator which calculates the position of the ground relative to the local NED height datum, started using range measurements that were close to zero - see plot below. Note - vehicle vertical position estimate NKF1.PD and terrain vertical position estimate NKF5.offset have had the sign flipped so that up is positive.

[image]

 This caused the terrain height to rise up to the vehicle height, which in turn made the EKF think it was close to ground so it started using the spurious range value. 

Looking at the code, one possibility is that something caused the EKF to think it was on ground  which will cause it to set the range to the default on-ground value if the range reading is outside limits. However I cannot find the log data to prove or disprove that hypothesis.

festlv, 2018-06-12:
> This caused the terrain height to rise up to the vehicle height, which in turn made the EKF think it was close to ground so it started using the spurious range value.
Shouldn't that be reflected in NKF5.offset? In the graph, rangefinder innovation is what's getting close to height estimate while offset stays about the same.

priseborough, 2018-06-12:
That happens later. Initially the terrain estimator innovation jumps, but the terrain estimate stays constant because the range data fails innovation checks. Eventually the terrain estimator accepts the measurement and the innovation jumps down and the terrain measurement jumps up. that is when the EKF starts using the range data as the height source.

festlv, 2018-06-12:
I added logging of onGround variable of NavEKF2_Core in Replay. The value is written to FIX field. It doesn't seem to change while in air (I added a plot of RC output to show when vehicle took off):

[image]

rijun034, 2018-11-05:
[https://drive.google.com/open?id=1LZyPhObXW30uKauDdFdQc6oHD91hzYSa](url)

Can anybody analyze this crash??
[image]

rmackay9, 2019-09-10:
I think we may have had another incident here: https://discuss.ardupilot.org/t/uncommanded-assent/46847/2