ArduPilot/ardupilot #7132: SITL instances limited to 256?

FranciscoFabra, 2017-10-26:
#### Issue details
First of all, thanks for a great job.
I am developing a multiUAV simulator and I have detected a scalability unexpected problem. I can not run more than 256 arducopter (SITL) instances because when I use a instance value equal or greater than 256 the instance tries to use ports starting from the initial 5760 port.
Is the instance identifier limited to 255 (8bits identifier)?
Is there any solution?

Thank you.

#### Version
3.5.0

#### Platform
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Embedded airframe into SITL (I think that it is an hexacopter)

#### Hardware type
SITL

#### Logs

jaxxzer, 2017-10-26:
I don't know if there is a limitation of SITL in particular, but the system id in the MAVLink spec is only 8 bits, so you are limited to 256 uniquely identifiable MAVs.

OXINARF, 2017-10-31:
Yes, it is a limitation in *instance* number: https://github.com/ArduPilot/ardupilot/blob/73a76455d2b69f8a8f1edadc5716edd546087b22/libraries/AP_HAL_SITL/SITL_State.h#L140

We can certainly increase it, but, as @jaxxzer said, you'll hit the MAVLink limit at 255 vehicles - I'm not sure if you are even changing the MAVLink ID though.

Another thing to be careful about is the ports, making sure they don't end up overlapping. With 256 instances I **think** it doesn't with the default ports.

FranciscoFabra, 2017-10-31:
Thank you for your help.
I am developing an upper logic layer over SITL, so the instance identifier is managed by my application without any limitation. The UAVs communicate through the main application and without using their inner identifier, in other words, they don't use MAVLink and their identifier for communications, so this limitation only afects to my application due to the port assignment.
I can state that the port for instance 256 is the same as the port for the instance 0 (port 5760), so I can not instantiate more than 256 SITL.
It would be great if you could specify the ports to be used whatever is the instance number, or if you could use a instance number of 16bits. Until then, my application will only be able to run 256 UAVs. It's not a tragedy, only a limitation. I only suggest this modification to improve new uses of SITL.
Thanks again.

WickedShell, 2019-06-27:
You should be able to use the `--base-port` option on the SITL binary to shift the ports around, but you'd need to be using an instance number within the 0 to 255 range.

closed 2019-06-27 by WickedShell