ArduPilot/ardupilot #7134: Change GPS Radius Accuracy

JoseOcasio, 2017-10-26:
#### Issue details
Feature: Ability to change GPS radius accuracy
Issue: Using GPS, the drone does not hover in place. We have seen oscillation of around 2 feet.

#### Version
3.6
#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Quad Frame. 

#### Hardware type
Pixhawk with Marvelmind Indoor "GPS"
#### Logs
N/A

rmackay9, 2017-10-27:
we should clear up the title a bit I think to avoid confusion.
This is an enhancement request to improve support for the MarvelMind in Copter?  Or is this a support question related to the drone not holding position well while using a regular GPS?

amilcarlucas, 2017-10-30:
@JoseOcasio We need an answer on this. The GPS accuracy can already be set on the EKF

JoseOcasio, 2017-10-31:
Hello Lucas. I am the same guy that put the email chain together with Marvelmind. This is regarding the same issue we are discussing in that thread.

OXINARF, 2017-10-31:
I'm closing this based on two reasons:

- @JoseOcasio is (apparently) using the MarvelMind GPS NMEA output instead of the recommend beacon mode; that means that the system is subject to the limitations of NMEA driver
- you can already change several GPS parameters in the EKF to make it thrust the GPS position more

Given the two points above, this is a support request and not an issue or feature request. As is known, support requests should be posted in the forum (http://discuss.ardupilot.org).

closed 2017-10-31 by OXINARF